cb22/macbook12-spi-driver #3: Caps Lock LED doesn't work

cgutman, 2016-09-16:
We probably need to capture the SPI transfers that happen on Windows and macOS to turn the LED on and off to make this work.

cb22, 2016-09-20:
I captured the packet needed to do this a while back, let me dig them up. If you'd like to experiment with sniffing the SPI data yourself, here's how I did it in Windows 10:

> It's possible to get Windows to dump everything it sends on what it calls SPB - basically SPI & I2C. Open the Event Viewer and enable View -> Show Analytic and Debug Logs. Then head to Application and Services Logs -> Microsoft -> Windows -> SPB-ClassExtension. Right click on Analytic and enable it - all events will be logged. Disable it when done, and look for the 1023 events, and view the details tab - they'll give you the full buffer contents. Other events will let you know things like the direction of the transfer, etc.

There are still quite a few packets with unknown contents. Definitely a cool one to reverse would be the "force click" that happens; but to the best of my knowledge that only happens under macOS.

roadrunner2, 2017-02-18:
Can you dig up those traces? I'd love to add this functionality (and submit a pull request).

cb22, 2017-02-21:
@roadrunner2 I don't think I have them anymore, but I'll do a capture from Windows. Sorry, I haven't had much time to give to this driver as of late

roadrunner2, 2017-04-14:
I got some traces from @tudorbarascu over on issue #6, and am making progress on this - the basics are working, but still have some reliability issues. Will send a pull request when it's ready.

closed 2017-05-13 by cb22