cb22/macbook12-spi-driver #7: Unclaimed $905 bounty for this

readnotify, 2017-01-22:
This might be solving the following bounty:-

https://www.bountysource.com/issues/35422234-macbook8-1-12-inch-early-2015-keyboard-and-trackpad-don-t-work

If so, someone should post the solution there and claim it?

In my case - I've got a Macbook Pro 2016 with touchbar, and the inbuilt keyboard/mouse don't work for me in Fedora.

roadrunner2, 2017-01-24:
Regarding the keyboard and touchpad on MBP 2016, I've been documenting the steps on a gist:

Dunedan, 2017-01-24:
Nice, I'm building a similar hardware support overview at: https://github.com/Dunedan/mbp-2016-linux

cb22, 2017-01-24:
Just a quick one about suspend / resume - I believe (if it's the same issue) it's related to the NVMe device not coming back out of sleep (see the off-topic conversation on the PR here https://github.com/cb22/macbook12-spi-driver/pull/2)

Perhaps try booting with `no_console_suspend=1` - it should suspend and then resume, but be stuck. Look for something like "device did not transition from D3" (I can't remember the exact error). Of course, this would only be the case if it's the same issue as on the MacBook9,1.

On the MacBook9,1 at least, it did mean that booting from a USB drive meant that suspend / resume actually worked.

roadrunner2, 2017-01-24:
@Dunedan Excellent! And I see you have a MBP13,2, and that that has the Apple NVMe controller - so I guess it's just the 13,3 that has the Samsung controller.

@cb22 Thanks for that info.

cb22, 2017-09-21:
Just a quick note on the bounty - I don't really feel comfortable claiming it for myself. 

I might have done the initial hacked together part, but without the contribution of everyone else involved (especially @l1k and @roadrunner2) this driver would be nowhere near where it is at the moment.

roadrunner2, 2017-09-22:
As I've noted elsewhere, I'm not interested in the bounty - I'm perfectly happy with you claiming it or splitting it with others.