cb22/macbook12-spi-driver #10: Touchpad is too sensitive

cb22, 2017-01-26:
The touchpad seems to detect touches even from a finger hovering slightly above it.

Maximus-, 2017-05-28:
I'm experiencing a mix of over-sensitivity and under-sensitivity. For example, when typing, it'll recognize my palm and move the mouse at the slightest touch. However, in terms of precision, if I attempt to move the cursor just a small amount, it's largely unresponsive, and then I end up moving it too far. 

Regarding the sensitivity for my palm, I know libinput/xorg/whatall have support for "palm recognition", however, it tends to just disable the trackpad while typing. Leaving this off, I've noticed I can't type while moving the cursor. I'm not sure if this is the drivers fault, or if it's libinput/X11.

mrodessit, 2018-03-27:
Any suggestion how to fix that sensivity? I even almost dont touch my touchpad.
Have macbook 12 (late 2017).

roadrunner2, 2018-04-02:
@mrodessit Grab [61-libinput-local.hwdb](https://gist.github.com/roadrunner2/1289542a748d9a104e7baec6a92f9cd7#file-61-libinput-local-hwdb) from my gist and put it in `/etc/udev/hwdb.d/61-libinput-local.hwdb` (then run `sudo systemd-hwdb update` and log out and back in, or reboot). The values there are for a MBP13,3 touchpad, so you might need to adjust them - `libinput-measure-touch-size` may be helpful here in figuring out what values to set.