cb22/macbook12-spi-driver #11: I need some help using this driver

xinyifly, 2017-03-05:
I'm using latest Arch Linux, while I try to use `iasl -d dsdt`, I found a `segmentation fault`.
So I turned to install `acpica`(aur) instead.

Still got a warning but I ignored it
```
Disassembly completed
ASL Output:    dsdt.dsl - 308873 bytes

 iASL Warning: There were 2 external control methods found during
 disassembly, but only 0 were resolved (2 unresolved).
```

I changed the !OSDW to OSDW at the correct place, then `iasl -tc dsdt.asl`
```
ASL Input:     dsdt.dsl - 10070 lines, 308872 bytes, 3791 keywords
AML Output:    dsdt.aml - 34877 bytes, 953 named objects, 2838 executable opcodes
Hex Dump:      dsdt.hex - 327305 bytes

Compilation complete. 0 Errors, 9 Warnings, 6 Remarks, 43 Optimizations
```

Warnings or Remarks are 1 of `Method Local is set but never used`,  `Reserved method should not return a value`, `Method Argument is never used`, `Control Method should be made Serialized`

I tried https://wiki.archlinux.org/index.php/DSDT#Using_a_CPIO_archive and got
```
[root@tux ~]# dmesg | grep -i dsdt
[    0.000000] ACPI: DSDT ACPI table found in initrd [kernel/firmware/acpi/dsdt.aml][0x883d]
[    0.000000] ACPI: DSDT 0x000000008AFEB000 008875 (v03 APPLE  MacBook  00080001 INTL 20140424)
[    0.639233] ACPI: \_SB_.PCI0.LPCB.EC__: Used as boot DSDT EC to handle transactions
[    0.839455] ACPI: \_SB_.PCI0.LPCB.EC__: Used as boot DSDT EC to handle transactions and events
```
but no clue for anything was override
```
[root@tux ~]# dmesg | grep -i override
[    0.000000] Command line: initrd=\acpi_override initrd=\initramfs-linux.img root=PARTUUID=e6323f91-bb51-4563-aa54-15d7f490dbb1 rw noapic intremap=nosid
[    0.000000] Kernel command line: initrd=\acpi_override initrd=\initramfs-linux.img root=PARTUUID=e6323f91-bb51-4563-aa54-15d7f490dbb1 rw noapic intremap=nosid
```

Then, I was failed to get my keyboard work after `make test` with this driver
```
make -C /lib/modules/4.9.11-1-ARCH/build M=/root/git/cb22/macbook12-spi-driver modules
make[1]: Entering directory '/usr/lib/modules/4.9.11-1-ARCH/build'
  CC [M]  /root/git/cb22/macbook12-spi-driver/applespi.o
  Building modules, stage 2.
  MODPOST 1 modules
  CC      /root/git/cb22/macbook12-spi-driver/applespi.mod.o
  LD [M]  /root/git/cb22/macbook12-spi-driver/applespi.ko
make[1]: Leaving directory '/usr/lib/modules/4.9.11-1-ARCH/build'
sync
rmmod applespi
insmod ./applespi.ko
```
keyboard still not working

I suspect something wrong with my DSDT override and tried recompile kernel with abs.
I modified this file and added these two lines.
```
[root@tux linux]# grep -i custom_dsdt config.x86_64 
# CONFIG_ACPI_CUSTOM_DSDT is not set
CONFIG_ACPI_CUSTOM_DSDT_FILE="/lib/firmware/dsdt.hex"
CONFIG_ACPI_CUSTOM_DSDT=y
```
I didn't touch config because I'm using an x64 linux
```
[root@tux linux]# grep -i custom_dsdt config
# CONFIG_ACPI_CUSTOM_DSDT is not set
```

I recompiled the kernel with `makepkg -s` and installed it with `pacman -U linux-4.9...` and `pacman -U linux-headers-4.9...` (It took me hours :( )

And my dsdt.hex
```
[root@tux linux]# ls /lib/firmware/dsdt.hex 
/lib/firmware/dsdt.hex
```

Some additional errors in dmesg
```
[root@tux linux]# dmesg | grep -i error
(Repeated ACPI Errors were deleted here)
[    0.523712] ACPI Error: Method parse/execution failed [\_PR.CPU1.APPT] (Node ffff8802660c8870), AE_ALREADY_EXISTS (20160831/psparse-543)
[    0.523740] ACPI Error: Method parse/execution failed [\_PR.CPU1.GCAP] (Node ffff8802660c8140), AE_ALREADY_EXISTS (20160831/psparse-543)
[    0.523759] ACPI Error: Method parse/execution failed [\_PR.CPU1._PDC] (Node ffff8802660c83e8), AE_ALREADY_EXISTS (20160831/psparse-543)
[    0.523776] ACPI: Marking method _PDC as Serialized because of AE_ALREADY_EXISTS error
[    0.524189] ACPI Error: [\_SB_.OSCP] Namespace lookup failure, AE_NOT_FOUND (20160831/psargs-359)
[    0.524204] ACPI Error: Method parse/execution failed [\_PR.CPU2.GCAP] (Node ffff8802660c8af0), AE_NOT_FOUND (20160831/psparse-543)
[    0.524222] ACPI Error: Method parse/execution failed [\_PR.CPU2._PDC] (Node ffff8802660c8de8), AE_NOT_FOUND (20160831/psparse-543)
[    0.524641] ACPI Error: [\_SB_.OSCP] Namespace lookup failure, AE_NOT_FOUND (20160831/psargs-359)
[    0.524656] ACPI Error: Method parse/execution failed [\_PR.CPU3.GCAP] (Node ffff8802660c80f0), AE_NOT_FOUND (20160831/psparse-543)
[    0.524674] ACPI Error: Method parse/execution failed [\_PR.CPU3._PDC] (Node ffff8802660c8ed8), AE_NOT_FOUND (20160831/psparse-543)
[    4.085433] intel-lpss: probe of 0000:00:15.0 failed with error -22
[    4.102090] intel-lpss: probe of 0000:00:15.2 failed with error -22
[    4.119016] intel-lpss: probe of 0000:00:19.0 failed with error -22
[    4.135292] hci_bcm: probe of BCM2E7C:00 failed with error -22
[    4.152127] intel-lpss: probe of 0000:00:1e.0 failed with error -22
[    4.215388] intel-lpss: probe of 0000:00:1e.3 failed with error -22
[    4.321937] brcmfmac 0000:02:00.0: Direct firmware load for brcm/brcmfmac4350c2-pcie.txt failed with error -2
```

```
[root@tux linux]# find /sys/devices -name *APP000D*
/sys/devices/pci0000:00/0000:00:1e.3/APP000D:00
/sys/devices/LNXSYSTM:00/LNXSYBUS:00/PNP0A08:00/device:19/APP000D:00
```

```
[root@tux linux]# lspci 
00:00.0 Host bridge: Intel Corporation Skylake Host Bridge/DRAM Registers (rev 08)
00:02.0 VGA compatible controller: Intel Corporation HD Graphics 515 (rev 07)
00:14.0 USB controller: Intel Corporation Sunrise Point-LP USB 3.0 xHCI Controller (rev 21)
00:15.0 Signal processing controller: Intel Corporation Sunrise Point-LP Serial IO I2C Controller #0 (rev 21)
00:15.2 Signal processing controller: Intel Corporation Sunrise Point-LP Serial IO I2C Controller #2 (rev 21)
00:16.0 Communication controller: Intel Corporation Sunrise Point-LP CSME HECI #1 (rev 21)
00:19.0 Signal processing controller: Intel Corporation Sunrise Point-LP Serial IO UART Controller #2 (rev 21)
00:1c.0 PCI bridge: Intel Corporation Device 9d10 (rev f1)
00:1d.0 PCI bridge: Intel Corporation Sunrise Point-LP PCI Express Root Port #9 (rev f1)
00:1d.1 PCI bridge: Intel Corporation Device 9d19 (rev f1)
00:1e.0 Signal processing controller: Intel Corporation Sunrise Point-LP Serial IO UART Controller #0 (rev 21)
00:1e.3 Signal processing controller: Intel Corporation Sunrise Point-LP Serial IO SPI Controller #1 (rev 21)
00:1f.0 ISA bridge: Intel Corporation Device 9d46 (rev 21)
00:1f.2 Memory controller: Intel Corporation Sunrise Point-LP PMC (rev 21)
00:1f.3 Audio device: Intel Corporation Sunrise Point-LP HD Audio (rev 21)
00:1f.4 SMBus: Intel Corporation Sunrise Point-LP SMBus (rev 21)
01:00.0 Mass storage controller: Apple Inc. Device 2003 (rev 11)
02:00.0 Network controller: Broadcom Limited BCM4350 802.11ac Wireless Network Adapter (rev 05)
03:00.0 Multimedia controller: Broadcom Limited 720p FaceTime HD Camera
```

What was wrong with my procedure or anything else was I supposed to do?

Link to StackExchange
http://unix.stackexchange.com/questions/349259/new-macbook-2016-april-12-inch-keyboard-problem-spi-and-dsdt

xinyifly, 2017-03-06:
Update: I increased OEM version in DSDT.dsl and got
```
[root@tux tmp]# dmesg | grep -i dsdt
[    0.000000] ACPI: DSDT ACPI table found in initrd [kernel/firmware/acpi/dsdt.aml][0x883d]
[    0.000000] ACPI: Table Upgrade: override [DSDT-APPLE - MacBook]
[    0.000000] ACPI: DSDT 0x000000008AFEB000 Physical table override, new table: 0x000000008A862000
[    0.000000] ACPI: DSDT 0x000000008A862000 00883D (v03 APPLE  MacBook  00080002 INTL 20160930)
[    0.617182] ACPI: \_SB_.PCI0.LPCB.EC__: Used as boot DSDT EC to handle transactions
[    0.827412] ACPI: \_SB_.PCI0.LPCB.EC__: Used as boot DSDT EC to handle transactions and events
```
But still can't get my keyboard work after type `make test`

xinyifly, 2017-03-06:
Is it because the Arch Kernel > 4.7 ? DSDT patch not working ? https://github.com/cb22/macbook12-spi-driver/pull/1

Dunedan, 2017-03-06:
I have it working here with Linux 4.10, so the kernel version shouldn't be the issue.

Have you already checked that the modification you did to the DSDT is there after you started up with it being patched?

xinyifly, 2017-03-07:
@Dunedan Thank you for help. I updated the description and checked the DSDT above.

Pamelloes, 2017-03-07:
I'm also having this same issue with 4.9.11-1-ARCH. I've patched the DSDT successfully (the override is logged in dmesg) and I've passed intremap=nosid to the kernel. However, I get the same dmesg output as xinyifly and the keyboard does not work.

omnibrain, 2017-03-15:
I'm having the same problem with 4.10.1-1-ARCH.

Pamelloes, 2017-03-15:
I was able to fix the issue by compiling the patches DSDT table into the kernel; the dynamic override wouldn't work.

omnibrain, 2017-03-16:
Still not working for me unfortunately... I'm getting the exact same outputs as posted above.

Pamelloes, 2017-03-16:
Also make sure you have "intremap=none" but NOT "noapic" as kernel parameters. Noapic will disable spi as a side effect.

omnibrain, 2017-03-16:
THANK YOU!!!! That was exactly the problem, I had the "noapi" kernel parameter. Now it's finally working.

PS you probably meant "intremap=*nosid*"

jnvandermeer, 2017-03-16:
... does this mean that keyboard and trackpad can be activated (and
everything is working)?

omnibrain, 2017-03-16:
I've only been using it for an hour or so but I'm very happy with it, absolutely no complaints so far. Great work @xinyifly !! You should put a bitcoin address in the readme for donations :)

Pamelloes, 2017-03-16:
(FYI the driver was written by @cb22)

omnibrain, 2017-03-16:
Oops you're right. Edited my comment.

xinyifly, 2017-03-17:
Cheers guys! The driver and the `noapic` information by @Pamelloes saved my life. @cb22 Should we close this issue now?

PS: An update with README for avoid using `noapic` kernel parameter would be helpful :)

omnibrain, 2017-03-20:
Sorry for the off-topic question but did you get suspend/hibernate working on the early 2016 macbook9,1?

Pamelloes, 2017-03-20:
No, I've spent a few days trying to get it to work; I've tried all sorts of horrible kernel hacks and have consulted with various channels on Freenode to no avail :( The one bit of progress I did make was setting the kernel parameter no_console_suspend=1 which revealed debug output from during suspend that was lost otherwise: it turns out that the issues is the hard drive doesn't get turned back on after any form of suspend. If you have any suggestions for how to try and fix this, or where a better forum for discussing the issue would be, let me know :)

omnibrain, 2017-03-20:
That's too bad... On the Arch Wiki it says:

> Suspend-to-RAM works out of the box.

...under the macbook9,1 model specific information. I guess that's just plain wrong then :(

omnibrain, 2017-03-20:
@carinawi if my stalking is correct, you are the author of the macbook9,1 article of the Arch wiki. Does/did suspend really work for you? Have you done anything to make it work or did it just work out of the box?

cb22, 2017-04-17:
@xinyifly 
Good to hear it's sorted; I'll change the README to make a note about not needing `noapic` (`intremap=nosid` is what replaces it)

@Pamelloes 
Yeah, the big problem with suspend / resume is currently the fact that the NVMe device disappeared. I did some digging on that which is documented in #2 - but unfortunately I never got around to testing sending custom opcodes.

In a nutshell, the NVMe device disappears, and refuses to be woken up. It seems like the Apple driver runs some custom commands on suspend / resume to ensure the device is in a usable state.

closed 2017-07-04 by cb22