cb22/macbook12-spi-driver #15: AUR Package not working with Macbook8,1

medismailben, 2017-04-24:
Hi,

First, I would like to thank you for your great work. This piece of code might be the only thing that allows people to use Linux on their new mac. 

I've tried installing the driver on my MacBook8,1 (12" 2015) as you explain it, and it didn't work.

Then I check the AUR repo and I was happy to find a distributed version on your driver called "macbook12-spi-driver-dkms". 

The installation ran smoothly and the driver is installed according to dkms status however the keyboard and the trackpad are not working.

Just to be clear, I'm running Archlinux with kernel 4.10.11-1 and the initremap=nosid is enabled.

Am I missing something ??

cb22, 2017-04-24:
Hey @medismailben 

You might want to read through #9 - it seems that on the 2015 model, the LPSS module needs to be disabled for things to work.

Furthemore, there's current a pull request (#14) that apparently significantly improves things on the Macbook8,1 as well as the reliability of initialization and the driver in general.

Maximus-, 2017-04-24:
This was the difference between things working and not working here on the macbook8,1. 

```
~/linux $ diff .config goodconfig
425c425
< CONFIG_X86_INTEL_LPSS=y
---
> # CONFIG_X86_INTEL_LPSS is not set
427,428c427
< CONFIG_IOSF_MBI=y
< # CONFIG_IOSF_MBI_DEBUG is not set
---
> # CONFIG_IOSF_MBI is not set
2146,2155d2144
< CONFIG_PINCTRL=y
< 
< #
< # Pin controllers
< #
< # CONFIG_DEBUG_PINCTRL is not set
< # CONFIG_PINCTRL_CHERRYVIEW is not set
< # CONFIG_PINCTRL_BROXTON is not set
< # CONFIG_PINCTRL_GEMINILAKE is not set
< # CONFIG_PINCTRL_SUNRISEPOINT is not set
3203c3192
< CONFIG_DW_DMAC_PCI=y
---
> # CONFIG_DW_DMAC_PCI is not set
```

christophgysin, 2017-04-27:
@medismailben I'm the author of that AUR package.

Now that #14 is merged, is this still an issue? If a module needs to be blacklisted, I could add it to dkms.conf so that it will be done automatically during installation.

Maximus-, 2017-04-28:
@christophgysin From my experience, even after displaying the LPSS module in the source tree, and recompiling, it won't work. I think there are some dependencies it has which are in-tree, and not modules, I just can't track them down. I went through all the dependencies that menuconfig showed me, but still couldn't get it. Just had to start with a fresh config. However, this is pure speculation.

medismailben, 2017-04-28:
@maximus

medismailben, 2017-04-28:
@Maximus- In #9, it seems that you made the keyboard work on 2015 12" MacBook, but now you say it won't work.

Also on Archlinux, I couldn't find the LPSS module. @christophgysin Even after pulling latest update of the package (version 0.35), the keyboard and trackpad are not detected.

I'm not sure if that a problem but I running the stock linux kernel (it is not recompiled).

Maximus-, 2017-04-28:
@medismailben  Ah, sorry if I'm unclear. The only way I've gotten things to work is by cloning a stock linux kernel and ensuring I don't enable LPSS or it's friends. I'll investigate some more in the upcoming weeks.