cb22/macbook12-spi-driver #25: can't run collectly

niwatolli3, 2017-05-13:
Hi, cb22-san,

Thank you for your driver.
I'm trying to use this driver, I have following problems.
* keyboard/trackpad are not working.
** There are the following log in dmesg when I press keyboard's key or click trackpad.
*** applespi spi-APP000D:00: 9600000 Hz actual, DMA

Do you know the meaning of the message?

niwatolli3, 2017-05-13:
Oh! I got it.

I added the following line into /etc/mkinitcpio.conf (https://gist.github.com/roadrunner2/1289542a748d9a104e7baec6a92f9cd7)
> applespi intel_lpss_pci spi_pxa2xx_platform appletb to the MODULES in /etc/mkinitcpio.conf.

But, why it is needed?

choelzl, 2017-05-13:
I try to install Arch on my Macbook2016 (12inch) i can't manage to get the keyboard to work ...

niwatolli3, 2017-05-15:
@choelzl 

Could you tell me what you do?
Did you compile kernel with modified DSDT table setting?

choelzl, 2017-05-15:
@niwatolli3 
I installed arch-linux with kernel 4.11, I downloaded kernel 4.11 and changed the config with allowing a modified dsdt, I placed my modified dsdt table to the location i specified in the config. I compiled and installed the modified kernel. 
The modified dsdt loads properly (detected in dmesg).
The keyboard shows up as a spi device, but the applespi driver doesnt seem to work or to attach itself to the keyboard. When I listen to events from the keyboar/trackpad i get nothing (when i press key it prints nothing).

niwatolli3, 2017-05-16:
@choelzl 
I see. I think you need to add the following things.
> applespi intel_lpss_pci spi_pxa2xx_platform appletb to the MODULES in /etc/mkinitcpio.conf.

niwatolli3, 2017-05-16:
@choelzl 
I met this situation, so I added the lines.
The problem was solved then I did it!

niwatolli3, 2017-05-16:
However, how did you install macbook12-spi-driver? using yaourt?
I installed it by using from download this repo & make & sudo make install.

choelzl, 2017-05-16:
I installed a kernel update, repatched a new kernel , installed the driver from the repo instead of yaourt, typed your commands and it seemed to work... Only problem is that sometimes when I boot it goes back to not working/picking up key pressses. Thanks

niwatolli3, 2017-05-17:
@choelzl 
That's a relief.
May I close this ticket?

choelzl, 2017-05-17:
Yes 👍

niwatolli3, 2017-05-17:
@choelzl 

Thanks!

closed 2017-05-17 by niwatolli3