cb22/macbook12-spi-driver #49: Keyboard/trackpad not responsive on resume for Macbook 8,1 (12" 2015)

rquast, 2017-10-24:
The keyboard and trackpad don't resume after suspend for me. I am using Ubuntu 17.04 and have tried this with both the 4.10 kernel and the mainline 4.14 rc5 kernel that both have the same issue. The issue appears to have affected other models in the past: https://github.com/cb22/macbook12-spi-driver/pull/30

roadrunner2, 2017-10-26:
Were there any related errors in `dmesg`? I presume the relevant driver messages show up? (`spi-device suspend done` and `spi-device resume done`)

rquast, 2017-10-26:
No error messages, both of those messages you mentioned appeared. Here's the output:

https://gist.github.com/rquast/519ade5d36e2eac73b20fbdf29a4fbe8

rquast, 2017-10-26:
Have tested to see if this worked after fixing https://github.com/cb22/macbook12-spi-driver/issues/50 but still doesn't resume.

roadrunner2, 2017-10-26:
Since there are no error messages, let's see what's going on with the initialization command. Just before suspending (assuming a working keyboard at this point), can you set the debug param to enable logging of some of the SPI packets:
```
echo 0x0501 | sudo tee /sys/module/applespi/parameters/debug
```
After resume you can turn off that logging by setting the debug param to 0 again. This should generate a moderate amount of output (a hexdump of each 256 byte packet, one packet for each key event plus a few others). Then provide the dmesg output for that, please.

rquast, 2017-10-26:
There's no debug output by the look. Is there something wrong there?

root@rquast-MacBook:/var/log# cat /sys/module/applespi/parameters/debug
1281

https://gist.github.com/rquast/8dac9b6bc946ee82f8e4b27d13f4aab1

roadrunner2, 2017-10-26:
That's very odd! Did you type anything on the (internal) keyboard after enabling the debug logs? At the very least before the suspend any such key events should've gotten logged.

rquast, 2017-10-26:
Yes, there's output when typing a few keys and so on before suspend, but nothing in the suspend itself.

https://gist.github.com/rquast/8559a388330ddecfd259f37160219e01

rquast, 2017-10-26:
Don't know if it helps, but the keyboard doesn't light up again after resume either. Also, no packets captured after resume.

roadrunner2, 2017-10-27:
Thanks for the info. Ok, I suspect we're running into a race here where a command is sent during shutdown but it's response is not received, and hence the write-outstanding flag is still set when we resume.

Can you do the following:
* change the debug flags to 0x507  (this will show any backlight and caps-lock-led commands sent too)
* Apply the following patch:
```patch
--- a/applespi.c
+++ b/applespi.c
@@ -1459,6 +1459,10 @@ static int applespi_resume(struct device *dev)
        struct applespi_data *applespi = spi_get_drvdata(spi);
        acpi_status status;
 
+       applespi->cmd_msg_queued = false;
+       applespi->write_active = false;
+       applespi->read_active = false;
+
        status = acpi_enable_gpe(NULL, applespi->gpe);
        if (ACPI_FAILURE(status)) {
                pr_err("Failed to re-enable GPE handler for GPE %d: %s\n",
```
Again, please show me the dmesg output (whether this patch helps or not).

rquast, 2017-10-27:
Applied the patch, but had to manually do that because it's different from master (wrong line number). Unfortunately, no change. https://gist.github.com/rquast/d5e970fb4891714fabdaf4c8839de279

rquast, 2017-10-27:
My apologies.. I didn't notice the change in the debug flag. Will re-run and paste the output.

rquast, 2017-10-27:
I don't think it gave any more information with the 0x507 flag: https://gist.github.com/rquast/b9dfbe14643729933fc0ba7852a9a663

roadrunner2, 2017-10-28:
Thanks. Sorry about the wrong line numbers - have some changes sitting in my tree and I figured `patch` would handle it.

So we're getting closer to the culprit. Can you apply this patch instead (the line numbers should be accurate now):
```patch
--- a/applespi.c
+++ b/applespi.c
@@ -674,6 +674,10 @@ applespi_send_cmd_msg(struct applespi_data *applespi)
        u16 crc;
        int sts;
 
+       pr_info("drain=%d cmd_msg_queued=%d init_cmd_idx=%d\n",
+               applespi->drain, applespi->cmd_msg_queued,
+               applespi->init_cmd_idx);
+
        /* check if draining */
        if (applespi->drain)
                return 0;
@@ -749,6 +753,7 @@ applespi_send_cmd_msg(struct applespi_data *applespi)
 
        sts = applespi_async(applespi, &applespi->wr_m,
                             applespi_async_write_complete);
+       pr_info("applespi_async: sts=%d\n", sts);
 
        if (sts != 0) {
                pr_warn("Error queueing async write to device: %d\n", sts);
```
And again show the dmesg output.

Right now it's looking like `spi_async` may actually be swallowing up the init request, indicating some issue with the keyboard/trackpad itself.

One reason for this could be an issue with the SIST ACPI call - try adding this patch too:
```patch
--- a/applespi.c
+++ b/applespi.c
@@ -617,9 +617,11 @@ static int applespi_enable_spi(struct applespi_data *applespi)
        long long unsigned int spi_status;
 
        /* Check if SPI is already enabled, so we can skip the delay below */
+       /*
        result = acpi_evaluate_integer(applespi->sist, NULL, NULL, &spi_status);
        if (ACPI_SUCCESS(result) && spi_status)
                return 0;
+       */
 
        /* SIEN(1) will enable SPI communication */
        result = acpi_execute_simple_method(applespi->sien, NULL, 1);
```

rquast, 2017-10-28:
No luck, but here's the output from the new info you are logging: https://gist.github.com/rquast/7b9cfe53cfe1a94bd926ac7722e18dc3

rquast, 2017-10-28:
Note: I don't have the earlier patch applied to this one. Worked from a fresh applespi.c with the new patches you provided.

roadrunner2, 2017-10-28:
Thanks - this basically confirms that the issue appears to be with the spi device or driver, rather than a logic issue in this applespi driver: we're queueing an init message, but never get a completion. So the spi device or driver is not processing the messages. I'm sorry, but this is going to take some more back and forth to figure out.

Here's another patch (I'll attach this one - I'm guessing that's easier to grab than copy pasting): 
[resume-dbg-1.patch.txt](https://github.com/cb22/macbook12-spi-driver/files/1424685/resume-dbg-1.patch.txt) This one will print out a bunch of flags (state) of the spi controller before and after queueing the init message on resume - trying to see if that gives us any clues as to why it's not sending the messages. Also, at this point setting the debug flag isn't necessary anymore.

rquast, 2017-10-29:
No this is great! Thanks for helping like this. I am more than happy to take the time to keep working at it.

Here is the output: https://gist.github.com/rquast/b8a579ce053776784cef76512c2a5eac

roadrunner2, 2017-11-09:
@rquast Sorry for the delay - things were busy. Thank you for the output. That's helpful - the controller state looks fine, as far as what was printed. Here's an expanded version of the patch which will print some more details, especially from the low-level driver: [resume-dbg-2.patch.txt](https://github.com/cb22/macbook12-spi-driver/files/1456041/resume-dbg-2.patch.txt) - same steps as for the last one.

shrijitsingh99, 2017-12-31:
Has this issue been fixed?
If not is there anyway i can help in debugging?

roadrunner2, 2017-12-31:
@shrijitsingh99 No, this issue is still open AFAIK. And yes, you could help by doing the following:
* grab and apply the `resume-dbg-2.patch.txt` above to the `applespi` driver - if you can do this on the latest driver code from [my repo](/roadrunner2/macbook12-spi-driver/) (that has several more patches which are currently sitting in pull request #56) that would be great
* build and load the new applespi module
* turn on debugging: `echo 0x507  | sudo tee /sys/module/applespi/parameters/debug`
* suspend and resume
* post the applespi dmesg output from resume

So far the issue is looking to be in the existing kernel code, rather than this driver, so the next steps will most likely involve patching the kernel proper to get more debug info from there.

shrijitsingh99, 2017-12-31:
Trying to build the patched applespi module give me this make error
https://gist.github.com/shrijitsingh99/b26e76aef3528c24ef6e073b6e4d379a

roadrunner2, 2017-12-31:
Apologies, should've tested the patch against the latest code. Here is an updated patch:
[resume-dbg-3.patch.txt](https://github.com/cb22/macbook12-spi-driver/files/1595445/resume-dbg-3.patch.txt)

Rather than building via dkms it might be easier to just build the module via `make` and load it via `sudo rmmod applespi ; sudo insmod applespi.ko` - but whatever works for you.

shrijitsingh99, 2017-12-31:
This is the output i get
https://gist.github.com/shrijitsingh99/640cf2a6dd54f0fc06792244cfe7b56c

roadrunner2, 2018-01-22:
@shrijitsingh99 Thanks for running this and getting the output. It confirms what I stated earlier: the spi message is being queued in the spi core, but for some reason the actual device-specific driver is not sending the message (we never get a completion callback). So it looks to me like the SPI controller device, or its driver, are not being woken up correctly.

@andy-shev I see you've done work on the `spi-pxa2xx` driver - do you have any suggestions or ideas here? FYI, the pci device is
```
00:15.4 Serial bus controller [0c80]: Intel Corporation Wildcat Point-LP Serial IO GSPI Controller #1 [8086:9ce6] (rev 03)
```
Also: the resume appears to work fine on other MacBook(Pro)'s (though it's not been widely tested, so there may be issues on some others too, but it does work fine on MBP13,3 with a `8086:a12a` SPI device), and the main difference between the MB8,1 and the others is that this one uses the `spi_pxa2xx_pci` driver whereas the others use the `intel_lpss_pci` driver.

andy-shev, 2018-01-24:
On Broadwell a different PCI driver is used, indeed. Though, the SPI host controller driver is still the same. We did a lot of self-test against SPI  in PIO/DMA modes, so, I doubt is anything wrong there. The symptoms looks like you  didn't get an interrupt on receiver side which might mean that SPI is working on MAC in half-duplex mode. I dunno if it's supported by SPI PXA2xx driver.

roadrunner2, 2019-01-13:
@TheChatty (or anybody else interested): I was just wondering if for some reason the keyboard's SPI interface needs to be forcefully re-enabled after resume on the MB8,1 (from my reading of the DSDT it should already be re-enabling the SPI interface on resume, but...). Can you try this quick patch:
```diff
--- a/applespi.c
+++ b/applespi.c
@@ -813,9 +813,11 @@ static int applespi_enable_spi(struct applespi_data *applespi)
        unsigned long long spi_status;
 
        /* check if SPI is already enabled, so we can skip the delay below */
+       /*
        result = acpi_evaluate_integer(applespi->sist, NULL, NULL, &spi_status);
        if (ACPI_SUCCESS(result) && spi_status)
                return 0;
+       */
 
        /* SIEN(1) will enable SPI communication */
        result = acpi_execute_simple_method(applespi->sien, NULL, 1);
```
If that doesn't work, then maybe this:
```diff
--- a/applespi.c
+++ b/applespi.c
@@ -813,11 +813,16 @@ static int applespi_enable_spi(struct applespi_data *applespi)
        unsigned long long spi_status;
 
        /* check if SPI is already enabled, so we can skip the delay below */
+       /*
        result = acpi_evaluate_integer(applespi->sist, NULL, NULL, &spi_status);
        if (ACPI_SUCCESS(result) && spi_status)
                return 0;
+       */
 
        /* SIEN(1) will enable SPI communication */
+       result = acpi_execute_simple_method(applespi->sien, NULL, 0);
+       if (ACPI_FAILURE(result))
+               pr_err("SIEN(0) failed: %s\n", acpi_format_exception(result));
        result = acpi_execute_simple_method(applespi->sien, NULL, 1);
        if (ACPI_FAILURE(result)) {
                pr_err("SIEN failed: %s\n", acpi_format_exception(result));
```

TheChatty, 2019-01-13:
@roadrunner2: I tested your first and second patch and later even added info output that SIEN(0|1) get called even but to no avail. After resume the SPI transfers still time out. `rmmod` and `modprobe` of applespi doesn't help either.

denisix, 2019-09-14:
the same issue for me, Ubuntu 19.04, kernel 4.16.0-041600-generic, after resume - keyboard & touch doesnt work, tried the patches, alse the patch with SIEN(0|1), tried even multiple times calling applespi_enable_spi() and added some timeouts before SIEN(1), but still no luck:

applied debug using `echo 0x0507 | sudo tee /sys/module/applespi/parameters/debug` and gathered the logs - dmesg, lspci -vvv, lsmod before (ok) and after(fail):
https://gist.github.com/denisix/dcd12689bcc01a6022427d3b848933f6

please let me know if you need any other information!
thanks!

roadrunner2, 2019-09-14:
@denisix Looking at the dmesg.fail log, it looks like you are unloading `applespi` and some of the pxa2xx modules before suspend and loading them again after resume? Can you tell me exactly which modules are being unloaded/loaded? Also, I'm a bit confused by the following: according to your `dmesg.fail` at least the `applespi`, `spi_pxa2xx_platform`, and `spi_pxa2xx_pci` modules are failing to load after resume, however your `lsmod.fail` shows them loaded - any idea what's going on?

denisix, 2019-09-16:
@roadrunner2 I experimented in many different ways and it was configured in `/etc/hibernate/common.conf`:
```UnloadModules snd_via82cxxx usb-ohci applespi spi_pxa2xx_platform spi_pxa2xx_pci intel_lpss_pci
UnloadAllModules no
UnloadBlacklistedModules no
LoadModules auto
LoadModulesFromFile /etc/modules
```
now I commented all these options, tried with the provided modifications of `applespi.c` but still no result (i'm suspending my system with `hibernate-ram`, and `pm-suspend` if that makes sense).

The other strange situation detected after system resume, when I tried to unload the module with `rmmod applespi` - command just freeze, from lsmod i see the following:
```# lsmod|grep applespi
applespi               28672  -1
```

attached logs:
[dmesg.rmmod.txt](https://github.com/cb22/macbook12-spi-driver/files/3615008/dmesg.rmmod.txt)
[lspci.fail.txt](https://github.com/cb22/macbook12-spi-driver/files/3615009/lspci.fail.txt)
[lsmod.fail.txt](https://github.com/cb22/macbook12-spi-driver/files/3615010/lsmod.fail.txt)
[dmesg.fail.txt](https://github.com/cb22/macbook12-spi-driver/files/3615011/dmesg.fail.txt)
[lspci.ok.txt](https://github.com/cb22/macbook12-spi-driver/files/3615012/lspci.ok.txt)
[lsmod.ok.txt](https://github.com/cb22/macbook12-spi-driver/files/3615013/lsmod.ok.txt)
[dmesg.ok.txt](https://github.com/cb22/macbook12-spi-driver/files/3615014/dmesg.ok.txt)

thank you!

denisix, 2019-11-04:
sorry, any update on this?
what can I do to help perform some debugging/investigations during the suspend/wakup procedures?

oevermod, 2021-08-04:
@roadrunner2 Hi! I tried your suggested patch with the current source:
```diff
diff --git a/drivers/input/keyboard/applespi.c b/drivers/input/keyboard/applespi.c
index eda1b23002b5..9e59dfa4f239 100644
--- a/drivers/input/keyboard/applespi.c
+++ b/drivers/input/keyboard/applespi.c
@@ -688,12 +688,20 @@ static int applespi_enable_spi(struct applespi_data *applespi)
        unsigned long long spi_status;

        /* check if SPI is already enabled, so we can skip the delay below */
+       /*
        acpi_sts = acpi_evaluate_integer(applespi->sist, NULL, NULL,
                                         &spi_status);
        if (ACPI_SUCCESS(acpi_sts) && spi_status)
                return 0;
+       */

        /* SIEN(1) will enable SPI communication */
+        acpi_sts = acpi_execute_simple_method(applespi->sien, NULL, 0);
+        if (ACPI_FAILURE(acpi_sts)) {
+               dev_err(&applespi->spi->dev, "SIEN failed: %s\n",
+                       acpi_format_exception(acpi_sts));
+               return -ENODEV;
+       }
        acpi_sts = acpi_execute_simple_method(applespi->sien, NULL, 1);
        if (ACPI_FAILURE(acpi_sts)) {
                dev_err(&applespi->spi->dev, "SIEN failed: %s\n",
```

Here is the dmesg output with and without applying the patch:
[dmesg_without_patch.txt](https://github.com/cb22/macbook12-spi-driver/files/6930941/dmesg_without_patch.txt)
[dmesg_with_patch.txt](https://github.com/cb22/macbook12-spi-driver/files/6930943/dmesg_with_patch.txt)

I also tried to turn on the debugging as suggested:
```sh
# echo 0x507 | tee /sys/module/applespi/parameters/debug
tee: /sys/module/applespi/parameters/debug: Permission denied
```
Is it still being done like that or is there another way?
What i did do is enabling the `CONFIG_SPI_DEBUG` kernel config option.

i also tried to apply the patch `resume-dbg-3.patch.txt` that you provided earlier:
[applespi_resume-dbg-3.patch.diff.txt](https://github.com/cb22/macbook12-spi-driver/files/6931184/applespi_resume-dbg-3.patch.diff.txt)
but cannot compile it:
```sh
$ make -j $(nproc) && make -j $(nproc) modules_install
  DESCEND objtool
  CALL    scripts/atomic/check-atomics.sh
  CALL    scripts/checksyscalls.sh
  CHK     include/generated/compile.h
  CC [M]  drivers/input/keyboard/applespi.o
In file included from ./include/linux/printk.h:7,
                 from ./include/linux/kernel.h:19,
                 from ./include/linux/list.h:9,
                 from ./include/linux/kobject.h:19,
                 from ./include/linux/of.h:17,
                 from ./include/linux/irqdomain.h:35,
                 from ./include/linux/acpi.h:13,
                 from drivers/input/keyboard/applespi.c:44:
drivers/input/keyboard/applespi.c: In function ‘applespi_send_cmd_msg’:
./include/linux/kern_levels.h:5:18: warning: format ‘%d’ expects argument of type ‘int’, but argument 3 has type ‘ktime_t’ {aka ‘long long int’} [-Wformat=]
    5 | #define KERN_SOH "\001"  /* ASCII Start Of Header */
      |                  ^~~~~~
./include/linux/kern_levels.h:14:19: note: in expansion of macro ‘KERN_SOH’
   14 | #define KERN_INFO KERN_SOH "6" /* informational */
      |                   ^~~~~~~~
./include/linux/printk.h:420:9: note: in expansion of macro ‘KERN_INFO’
  420 |  printk(KERN_INFO pr_fmt(fmt), ##__VA_ARGS__)
      |         ^~~~~~~~~
drivers/input/keyboard/applespi.c:771:2: note: in expansion of macro ‘pr_info’
  771 |  pr_info("drain=%d cmd_msg_queued=%d\n",
      |  ^~~~~~~
drivers/input/keyboard/applespi.c: At top level:
drivers/input/keyboard/applespi.c:1925:10: fatal error: spi-pxa2xx.h: No such file or directory
 1925 | #include "spi-pxa2xx.h"
      |          ^~~~~~~~~~~~~~
compilation terminated.
make[3]: *** [scripts/Makefile.build:271: drivers/input/keyboard/applespi.o] Error 1
make[2]: *** [scripts/Makefile.build:514: drivers/input/keyboard] Error 2
make[1]: *** [scripts/Makefile.build:514: drivers/input] Error 2
make[1]: *** Waiting for unfinished jobs....
make: *** [Makefile:1841: drivers] Error 2
```

Unsure why it cannot find the `spi-pxa2xx.h` header file, maybe you have an idea?

I would be happy to assist in any way needed to get this to work on macbook 8,1

andy-shev, 2021-08-05:
> i also tried to apply the patch `resume-dbg-3.patch.txt` that you provided earlier:
> [applespi_resume-dbg-3.patch.diff.txt](https://github.com/cb22/macbook12-spi-driver/files/6931184/applespi_resume-dbg-3.patch.diff.txt)
> but cannot compile it:

> ```shell
> drivers/input/keyboard/applespi.c:1925:10: fatal error: spi-pxa2xx.h: No such file or directory
>  1925 | #include "spi-pxa2xx.h"
> ```
> 
> Unsure why it cannot find the `spi-pxa2xx.h` header file, maybe you have an idea?

In newer kernels the header has been moved under spi/ folder IIRC.

oevermod, 2021-08-06:
Thanks! Have managed to compile it now with the following changes:
[applespi_resume-dbg-3.patch.diff.txt](https://github.com/cb22/macbook12-spi-driver/files/6945068/applespi_resume-dbg-3.patch.diff.txt)

Here is the dmesg log from suspend and resume:
[dmesg_with_dbg_patch.txt](https://github.com/cb22/macbook12-spi-driver/files/6945073/dmesg_with_dbg_patch.txt)

Maybe it already provides some insight.

The third pr_info block sadly does not compile because of the following two errors:
```
error: ‘struct driver_data’ has no member named ‘cur_transfer’
error: ‘struct driver_data’ has no member named ‘pump_transfers’
```
I guess the pxa-2xx code has changed in the meantime, but i am really new to kernel and C development so i have not been able to figure out how to fix it.

andy-shev, 2021-08-13:
> Thanks! Have managed to compile it now with the following changes:
> [applespi_resume-dbg-3.patch.diff.txt](https://github.com/cb22/macbook12-spi-driver/files/6945068/applespi_resume-dbg-3.patch.diff.txt)
> 
> Here is the dmesg log from suspend and resume:
> [dmesg_with_dbg_patch.txt](https://github.com/cb22/macbook12-spi-driver/files/6945073/dmesg_with_dbg_patch.txt)
> 
> Maybe it already provides some insight.
> 
> The third pr_info block sadly does not compile because of the following two errors:
> 
> ```
> error: ‘struct driver_data’ has no member named ‘cur_transfer’
> error: ‘struct driver_data’ has no member named ‘pump_transfers’
> ```
> 
> full log: [applespi_compile_log.txt](https://github.com/cb22/macbook12-spi-driver/files/6945215/applespi_compile_log.txt)
> 
> I guess the pxa-2xx code has changed in the meantime, but i am really new to kernel and C development so i have not been able to figure out how to fix it.

Yes. code changed a lot lately.
Nevertheless, if you really want to debug SPI transfers, you need to enable SPI trace events (/sys/kernel/trace/events/spi/enable IIRC, but you may google for it) and collect corresponding events during the driver's resume/suspend cycle. Note, you may also enable / disable events programmatically from SPI client driver.

anomaly256, 2021-12-09:
Hi, are there any known workarounds for this issue currently?

denisix, 2022-07-10:
Hi, any ideas on this issue? Maybe we need to setup bounty for this issue? :)

anomaly256, 2022-12-30:
Still not working after resume in current kernels 😞

muojp, 2023-01-07:
I'm posting my personal workaround here; Turn off suspend/resume feature and make the OS automatically shutdown if the laptop is folded in battery mode for a long time.

It suits my use-case because I just want to prevent my Mac from draining too much battery on "suspended" state. Of course this workaround makes you lose all unsaved documents but it doesn't matter for me because I mostly use VSCode for editing texts and the editor is robust for sudden shutdown.

## Add an user to perform auto-shutdown

```
useradd autoshutdown
```

## Put an auto-shutdown script

```
#!/bin/bash

STATUS='closed'
STATUS_POWER='0'
FILE="/opt/autoshutdown/closed_count"
THRESHOLD=30

IS_CLOSED=$(grep $STATUS /proc/acpi/button/lid/LID0/state)
IS_POWER_DISCONNECTED=$(grep $STATUS_POWER /sys/class/power_supply/ADP1/online)

function reset() {
	echo -n 0 > $FILE
}

if [ "$IS_CLOSED" != '' ] && [ "$IS_POWER_DISCONNECTED" != '' ]; then
	NUM=$(cat $FILE)
	((NUM=NUM+1))
	if [[ $NUM -ge $THRESHOLD ]]; then
		reset
		sync ; sync
		sudo poweroff
	else
		echo -n $NUM > $FILE
	fi
else
		reset
fi
```

Saved as `/opt/autoshutdown/shutdown-by-continuous-lid-close.sh` and make it invoked every minute using cron.

## Set-up a cron entry

```
sudo crontab -u autoshutdown -e
```

```
* * * * * /opt/autoshutdown/shutdown-by-continuous-lid-close.sh
```

## Make it enable for cron-invoked processes to call `poweroff`

`/etc/sudoers.d/autoshutdown`

```
autoshutdown ALL=NOPASSWD: /sbin/poweroff
```

muojp, 2023-01-07:
By the way, do you folks managed to use Bluetooth on Macbook8,1?
I'm stuck in UART Bluetooth chip for several months: https://github.com/leifliddy/macbook12-bluetooth-driver/issues/18

skyostil, 2024-04-13:
Just to offer another workaround, you can use suspend-to-idle (s2idle, aka. S0ix) instead of deep suspend (S3) to avoid the broken keyboard and touchpad on resume. s2idle keeps more of the system running, so the power draw is higher, but at least you can use your computer afterwards :)

On my macbook8,1 the difference is about 1.1% battery drain per hour with deep suspend and 1.9% with s2idle.

To switch to s2idle, run this command and reboot:

`echo 'w /sys/power/mem_sleep - - - - s2idle' | sudo tee /etc/tmpfiles.d/sleep-mode.conf`

(To undo the change, `sudo rm /etc/tmpfiles.d/sleep-mode.conf`)

skyostil, 2024-04-13:
FWIW, I spent some time fiddling around with various ACPI commands and insmod/rmmod sequences, but nothing seems to bring the SPI channel back from the dead after an S3 resume. Any other ideas?

tvpittner, 2026-11-05:
I think I found the root cause of this, and have a working fix for S3 on a MacBook8,1 (Fedora 45, kernel 7.2.15).

**Cause:** after S3 resume, the Apple firmware does not re-initialise the private LPSS registers of the GSPI controller (`8086:9ce6`, `00:15.4`). After resume I read:

```
BAR0+0x800  PRV_CLOCK  = 0x00000000   (clock gated)
BAR0+0x804  PRV_RESETS = 0x00000000   (held in reset)
```

`spi-pxa2xx-pci` never writes these in PCI mode, so the controller has no clock, raises no interrupts (the IRQ counter in `/proc/interrupts` stays frozen), and every transfer times out. This matches what @roadrunner2 and @andy-shev saw above: messages get queued but never complete. It also explains why rmmod/modprobe, `SIEN`, and even a PCI remove/rescan don't help: none of them touch these two registers. Keyboard power (GPIO 13) and the SPI pin mux (GPIO 87–90) are fine after resume.

**Fix:** with the driver unbound, deassert reset (`0x804 = 0x3`) and enable the clock (`0x800 |= 1`), then bind again. That's the same thing coreboot does for Lynx Point/Broadwell SerialIO in PCI mode. This brings the keyboard and trackpad back, even live on an already-broken system without a reboot.

I packaged it as a `systemd-sleep` hook (unbind before S3, re-init + rebind after resume), with an explanation and a manual recovery recipe here:
https://github.com/tvpittner/macbook8-1-fedora/tree/main/s3-keyboard-fix

Tested on one machine so far (live fix, several `rtcwake` S3 cycles, and a 9.5-hour overnight lid-closed suspend; drain about 0.5 %/h vs. the ~1.9 %/h measured for s2idle above). Reports from other MacBook8,1 owners would be very welcome. A proper fix probably belongs in `spi-pxa2xx-pci` (restore `PRV_CLOCK`/`PRV_RESETS` on resume for LPT/WPT in PCI mode). Happy to help test a kernel patch.

Thanks to everyone in this thread. The earlier debugging here pointed the way. (Disclosure: the analysis and scripts were done together with Claude, Anthropic's AI model, and tested on my machine.)

tvpittner, 2026-11-05:
Small refinement to the fix above. I've now dumped the registers right after a **cold boot** too:

```
cold boot:  PRV_CLOCK (0x800) = 0x80190005   PRV_RESETS (0x804) = 0x00000003
after S3:   PRV_CLOCK (0x800) = 0x00000000   PRV_RESETS (0x804) = 0x00000000
```

So besides the enable bit, the firmware also programs the M/N clock divider (M = 2, N = 25, plus the update bit 31). Just setting `0x800 |= 1` (what I described above) works too, but the controller then may not get the clock the firmware intended. The hook now saves `PRV_CLOCK` before suspend and writes that exact value back after resume. This has been tested with an S3 cycle, and the keyboard and trackpad came back normally. The README and the manual recovery recipe are updated:
https://github.com/tvpittner/macbook8-1-fedora/tree/main/s3-keyboard-fix

ozhankaraman, 2026-11-05:
I will probably test this on weekend and keep you updated

andy-shev, 2026-11-13:
There was some activity in upstream about this I believe: https://lore.kernel.org/linux-spi/20260930160629.1822-1-fourdollars@debian.org/. But it seems abandoned. Can anybody to follow and we may find a solution in the kernel?

geeknees, 2026-11-14:
Thanks @tvpittner, @andy-shev, and everyone who investigated this over the years. Here is an **independent MacBook8,1 S3 resume hardware result** that may help with the in-kernel fix.

**Hardware and setup**
- MacBook8,1 (12-inch Early 2015, Core M-5Y31)
- Omarchy edge, kernel `7.2.15-2-omarchy`
- In-tree `applespi`, with `initcall_blacklist=dw_pci_driver_init` to force SPI PIO

**S3 validation**

I tested an opt-in `systemd-sleep` hook that saves and restores the LPSS private MMIO registers for `0000:00:15.4`, with SPI driver unbind/rebind.

Before suspend, the saved state includes:

- `PRV_CLOCK = 0x80190005`
- `PRV_RESETS = 0x00000003`

These values agree with @tvpittner's findings.

The journal confirms **three successful deep/S3 suspend-resume cycles**, including approximately **8 hours 14 minutes** and **27 hours 32 minutes**.

Each cycle records:

```text
S3 LPSS context saved
PM: suspend entry (deep)
PM: suspend exit
LPSS restore 0000:00:15.4
applespi OK after deep S3
```

The built-in keyboard and trackpad are functional after resume.

**Implementation**

https://github.com/geeknees/omarchy/tree/macbook81/poc-2026-11/install/hardware/apple/macbook81

This is a userspace workaround, **not a validation of the proposed in-kernel patch**.

I also noticed the [[v17 Linux SPI patch series](https://lore.kernel.org/linux-spi/20260930160629.1822-1-fourdollars@debian.org/)](https://lore.kernel.org/linux-spi/20260930160629.1822-1-fourdollars@debian.org/) referenced above. The LPSS register restoration work addresses the same underlying problem.

A proper kernel-side fix would be preferable to maintaining a distro-specific sleep hook.

I'm happy to provide `lspci -vv` output, additional logs, or test an updated kernel patch on another MacBook8,1 if that would help move the upstream work forward.

Thanks again for all the investigations that made this possible.