cb22/macbook12-spi-driver #50: Keyboard/trackpad not available during initramfs Macbook 8,1

rquast, 2017-10-24:
The keyboard is not available when needing to input a password for LUKS on Macbook 8,1. The keyboard is available before (in grub) and after initramfs. Here is a previous discussion on the issue: https://github.com/cb22/macbook12-spi-driver/issues/9#issuecomment-337811267

rquast, 2017-10-24:
Interesting thing I also found - if I remove all the suggested modules for /etc/initramfs-tools/modules then I no longer need to use irqpoll for the 4.14 rc5 kernel.

roadrunner2, 2017-10-26:
@rquast I believe you have the following SPI device:
```
00:15.4 Serial bus controller [0c80]: Intel Corporation Wildcat Point-LP Serial IO GSPI Controller #1 [8086:9ce6] (rev 03)
```
The driver for this is `spi-pxa2xx-pci`. So try adding that to your `/etc/initramfs-tools/modules` (and obviously with the `applespi` driver too - not sure about the `spi-pxa2xx-platform` now - I may well have gotten things confused with all the issues and different models etc that we've been debugging).

Your comment about not needing irqpoll if you remove all those modules is interesting! If you can narrow down the culprit further, that would be helpful so we can update the instructions.

rquast, 2017-10-26:
It appears that was the issue. Thanks a lot! :+1:

In addition to the modules you also had in the documentation, I added spi_pxa2xx_pci then did an initramfs update and a depmod then reboot. Dropped to an initramfs shell and the keyboard works.

It may be worth updating the documentation for the MacBook 8,1 12"?

rquast, 2017-10-26:
I also don't need to use irqpoll since this started working.

closed 2017-10-26 by rquast

roadrunner2, 2017-10-26:
Great! Yes, will update the README shortly. One question: is the `spi-pxa2xx-platform` module actually needed in the initramfs?

rquast, 2017-10-26:
Yes, have just tested and needs both the platform one as well as the PCI to work. Here are the required modules (doesn't need lpss one).

spi_pxa2xx_platform
spi_pxa2xx_pci
applespi

Thanks again!

JensKoehlerMW, 2017-11-15:
How can I integrate this into a live ISO? I already compiled applespi.ko? Btw, instead of "apt install dkms" I did "dnf install dkms", but I could not "dnf install initramfs-tools"?

TheChatty, 2017-11-15:
How can I integrate this into a live ISO? I already compiled applespi.ko? Btw, instead of "apt install dkms" I did "dnf install dkms", but I could not "dnf install initramfs-tools"?

roadrunner2, 2017-11-16:
@TheChatty You don't say, but it sounds like you might be using Fedora? If so, Fedora uses `dracut`, not `initramfs` - see [my gist](https://gist.github.com/roadrunner2/1289542a748d9a104e7baec6a92f9cd7#keyboardtouchpadtouchbar) for instructions for Fedora.

TheChatty, 2017-11-16:
You're correct. I use Fedora-Workstation-Live-x86_64-Rawhide-20171114.

I did:
dnf install dkms
git clone ...; dkms install applespi/0.1
dmesg|tail

[  733.918887] applespi: loading out-of-tree module taints kernel.
[  733.918939] applespi: module verification failed: signature and/or required key missing - tainting kernel
[  733.973449] input: Apple SPI Keyboard as /devices/pci0000:00/0000:00:15.4/pxa2xx-spi.0/spi_master/spi0/spi-APP000D:00/input/input13
[  733.976129] input: Apple SPI Touchpad as /devices/pci0000:00/0000:00:15.4/pxa2xx-spi.0/spi_master/spi0/spi-APP000D:00/input/input14
[  733.977786] applespi: spi-device probe done: spi-APP000D:00
[  734.845374] irq 21: nobody cared (try booting with the "irqpoll" option)
[  734.845382] CPU: 2 PID: 0 Comm: swapper/2 Tainted: G        W  OE   4.15.0-0.rc0.git1.1.fc28.x86_64 #1
[  734.845384] Hardware name: Apple Inc. MacBook8,1/Mac-BE0E8AC46FE800CC, BIOS MB81.88Z.0168.B00.1708080033 08/08/2017
[  734.845385] Call Trace:
[  734.845388]  <IRQ>
[  734.845394]  dump_stack+0x8e/0xd6
[  734.845399]  __report_bad_irq+0x35/0xc0
[  734.845405]  note_interrupt+0x24b/0x2a0
[  734.845410]  handle_irq_event_percpu+0x54/0x80
[  734.845415]  handle_irq_event+0x39/0x60
[  734.845420]  handle_fasteoi_irq+0x75/0x130
[  734.845425]  handle_irq+0xa9/0x110
[  734.845429]  do_IRQ+0x58/0x110
[  734.845434]  common_interrupt+0xa7/0xa7
[  734.845437]  </IRQ>
[  734.845440] RIP: 0010:cpuidle_enter_state+0x135/0x390
[  734.845442] RSP: 0018:ffffbb51c0cdbe70 EFLAGS: 00000202 ORIG_RAX: ffffffffffffff3e
[  734.845446] RAX: ffff90716c5b8000 RBX: 000000ab18327b49 RCX: 0000000000000000
[  734.845448] RDX: ffff90716c5b8000 RSI: 0000000000000001 RDI: ffff90716c5b8000
[  734.845449] RBP: ffffbb51c0cdbeb0 R08: 00000000ffffffff R09: 0000000000000000
[  734.845451] R10: 0000000000000000 R11: 0000000000000000 R12: ffff90716e3e5e00
[  734.845453] R13: 0000000000000000 R14: 0000000000000001 R15: ffffffffa3063918
[  734.845469]  cpuidle_enter+0x17/0x20
[  734.845472]  call_cpuidle+0x23/0x40
[  734.845475]  do_idle+0x186/0x1e0
[  734.845479]  cpu_startup_entry+0x73/0x80
[  734.845483]  start_secondary+0x181/0x1c0
[  734.845488]  secondary_startup_64+0xa5/0xa5
[  734.845499] handlers:
[  734.845557] [<ffffffffc0345400>] ssp_int [spi_pxa2xx_platform]
[  734.845606] [<ffffffffa261c010>] serial8250_interrupt
[  734.845610] Disabling IRQ #21
[  787.564700] IPv6: ADDRCONF(NETDEV_UP): wlp1s0: link is not ready

lsmod|grep 'spi\|lpss\|appletb'
intel_lpss_pci         20480  0
intel_lpss             16384  1 intel_lpss_pci
applespi               24576  0
spi_pxa2xx_platform    24576  0
spi_pxa2xx_pci         16384  0
8250_lpss              16384  0

But neither the keyboard nor the touchpad are working afterwards.

roadrunner2, 2017-11-19:
@TheChatty That 'irqpoll' error is probably the issue - it appears that on MB8,1 you need to boot with the applespi (and related) drivers present in the initrd - did you try rebooting after the above? If it still doesn't work, try adding `irqpoll` to your kernel command line.

TheChatty, 2017-11-19:
Now I upgraded to Fedora-Workstation-Live-x86_64-Rawhide-20171118 (because older version's kernel header were not available anymore) and did the same again (no irqpoll) and keyboard and touchpad are working - even keyboard light is on.

But dmesg after boot looks like:

```
[ 2523.738991] applespi: loading out-of-tree module taints kernel.
[ 2523.739045] applespi: module verification failed: signature and/or required key missing - tainting kernel
[ 2523.792529] input: Apple SPI Keyboard as /devices/pci0000:00/0000:00:15.4/pxa2xx-spi.0/spi_master/spi0/spi-APP000D:00/input/input13
[ 2523.793614] input: Apple SPI Touchpad as /devices/pci0000:00/0000:00:15.4/pxa2xx-spi.0/spi_master/spi0/spi-APP000D:00/input/input14
[ 2523.794724] applespi: spi-device probe done: spi-APP000D:00
[ 2523.800960] hpet_rtc_timer_reinit: 36 callbacks suppressed
[ 2523.800962] hpet1: lost 7134 rtc interrupts
[ 2523.805264] applespi: modeswitch done.
[ 2900.669473] hpet1: lost 9600 rtc interrupts
[ 2956.097253] hpet1: lost 24 rtc interrupts
[ 2956.281571] hpet1: lost 10 rtc interrupts
[ 2957.188924] hpet1: lost 56 rtc interrupts
[ 2957.993349] hpet1: lost 49 rtc interrupts
[ 2958.768245] hpet1: lost 48 rtc interrupts
[ 2958.972207] hpet1: lost 11 rtc interrupts
[ 2960.304323] hpet1: lost 83 rtc interrupts
[ 2960.364426] hpet1: lost 2 rtc interrupts
```

And many more lines of lost interrupts.