cb22/macbook12-spi-driver #52: Macbook 2017 12" ssd512 i5u 8gb usb-c all work fine

reproteq, 2017-10-27:
Macbook 2017 12" ssd512 i5u 8gb usb-c 
KALI LINUX 2017.2  INSIDE 
kernel 4.13
refind  
kali linux 2017.2 
cb22/macbook12-spi-driver

https://youtu.be/gHMjfi38ji0

rquast, 2017-10-28:
Does suspend/resume work properly with the 2017 12"? That is, does the keyboard still work after resume?

reproteq, 2017-10-28:
uff not man litle bug , after suspend resume problems errors and need reboot
other work good fn f1 f2 etc ...
solution fast 
disable supend mode

rquast, 2017-10-28:
it might be this... https://github.com/cb22/macbook12-spi-driver/issues/49

roadrunner2, 2017-10-28:
@reproteq Sorry, I'm a little unclear as to the state of resume on this laptop: you're saying that resume is not working, right? Can you be a bit more specific: what isn't working? Is the keyboard working or not? What else is not working?

reproteq, 2017-10-29:
all work fine keybord, mouse but when resume/suspend not working more i need reboot
is posible solution for this bug?

roadrunner2, 2017-11-09:
@reproteq Just before suspending (assuming a working keyboard at this point), can you set the debug param to enable logging of some of the SPI packets:
```sh
echo 0x0507 | sudo tee /sys/module/applespi/parameters/debug
```
After resume you can turn off that logging by setting the debug param to 0 again. This should generate a moderate amount of output (a hexdump of each 256 byte packet, one packet for each key event plus a few others). Then provide the dmesg output for that, please.