cb22/macbook12-spi-driver #55: Use GPIO and write a HID driver

joelkraehemann, 2017-12-13:
Hi,
I am not happy with your driver. The driver does generate newlines during boot and `modprobe -rv applespi && modprobe -v applespi`
Further there are GPIOs available. Might be you can use it to do a HID driver.

I have not much experience doing kernel drivers. However here are some thoughts done so far:
https://github.com/joelkraehemann/apple-spi/blob/master/gpio_applehidkbd.c

roadrunner2, 2017-12-13:
Can you elaborate on "does generate newlines..."? Sorry, but I don't really know what you're talking about here.

Regarding GPIO's - how is that supposed to work? Are you sure the SPI bus is running over some GPIO pins? And wouldn't that entail re-implementing the SPI protocol in the driver? Sorry, I'm just a bit confused.

l1k, 2017-12-18:
@joelkraehemann:

As stated in <a href="http://elixir.free-electrons.com/linux/latest/source/drivers/spi/spi-gpio.c">spi-gpio.c</a> (which you're including):
<blockquote>
This bitbanging SPI master driver should help make systems usable when a native hardware SPI engine is not available, perhaps because its driver isn't yet working or because the I/O pins it requires are used for other purposes.
</blockquote>

However in this case, an SPI master driver is readily available and supported by the Intel folks, so there's no need to reimplement or duplicate its functionality and a driver doing so wouldn't seem to be acceptable for mainline.

Moreover, so far there's only a single slave attached to the master, but what if Apple adds further slaves in future machines? Bitbanging would seem fragile in that case.

closed 2018-03-14 by cb22