cb22/macbook12-spi-driver #59: Change fn/ctrl/alt/super order

Anthony25, 2018-02-21:
Hi,
This [patched hid-apple module](https://github.com/free5lot/hid-apple-patched) has some options to change the order of the fn, ctrl, alt and super keys, to act like a PC keyboard. Is this possible for you to implement it on this spi driver?

Thanks a lot!

roadrunner2, 2018-02-26:
Hmm, generally I don't think such key remappings belong in the drivers - that's what the settings in Gnome, KDE, xkb, keymaps, etc are for (without then needing to duplicate the options for every single keyboard driver).

Having said that, the FN key is special, since that's handled in the driver directly, so I could see a case for a config option to allow just that key to be swapped with any of the other meta keys. @cb22 any thoughts/opinions?

omnibrain, 2018-02-26:
I would love the option to remap the FN key! My plan was to remap all the media keys on the function keys but configuring the FN key directly on the driver would be much more elegant

roadrunner2, 2018-05-20:
I've pushed a simple patch over on my [repo](/roadrunner2/macbook12-spi-driver/) that adds a new `fnremap` module parameter: it allows you swap the fn key with any of the 7 modifier keys (see the output from `modinfo applespi.ko` or the source code for valid values). See if this works for you.

P.S. when/if you do try it out, could you also report the output from `dmesg | grep "applespi: info"` over at #64 (see also Dunedan/mbp-2016-linux#61). TIA.

Anthony25, 2018-05-21:
@roadrunner2 thanks a lot, I'll test it tomorrow ;)

roadrunner2, 2018-08-10:
Anybody been able to test this yet?

Anthony25, 2018-08-10:
I tested it in June, but it wasn't working for me on a mbp 14.3. I'm sorry, I swapped this laptop with a thinkpad, so I cannot help you to debug more :(

roadrunner2, 2018-08-24:
Thanks for looking at this. A bit unfortunate that you didn't let me know back then so we could've debugged it. Oh well.

Anybody else try this? @omnibrain or @arno01?

roadrunner2, 2018-09-05:
@Dunedan, do you perchance have a couple spare cycles to test this? Sorry to bug you with this, but I'd like to include this in my next pull request and when upstreaming the driver, and for that I need at least one person to confirm it works.

Dunedan, 2018-09-13:
Sorry that took a while. I did a quick test and it seems to work fine. :+1:
As I'm not going to use this feature, I can't tell anything about possible side effects when using it a while.

arno01, 2018-09-17:
@roadrunner2 oh, I am not sure how did I miss that feature, but this is something that I really needed.
I've just switched to use it ``modprobe applespi fnremap=1`` and it is working like fine, much thanks! :-)

roadrunner2, 2018-09-18:
Thanks to both of you for testing this! Now I have a little more confidence in submitting it upstream :smile: