cb22/macbook12-spi-driver #60: Compilation error: "acpi_finish_gpe" undefined

zhovner, 2018-02-26:
I building module with custom kernel `4.14.22` and get the errors:

```
make: Entering directory '/home/gay/linux-4.14.22'
  Building modules, stage 2.
  MODPOST 1 modules
WARNING: "acpi_finish_gpe" [../macbook12-spi-driver//applespi.ko] undefined!
WARNING: "acpi_enable_gpe" [../macbook12-spi-driver//applespi.ko] undefined!
WARNING: "acpi_disable_gpe" [../macbook12-spi-driver//applespi.ko] undefined!
  CC      ../macbook12-spi-driver//applespi.mod.o
  LD [M]  ../macbook12-spi-driver//applespi.ko
make: Leaving directory '/home/gay/linux-4.14.22'
```

Here is my kernel [.config](https://pastebin.com/nM2DHbEN) file.

zhovner, 2018-02-26:
I found that `APPLE_GMUX=m` is required. Fixed now.

closed 2018-02-26 by zhovner

roadrunner2, 2018-03-01:
I think your problem was actually that you didn't have `ACPI=y` - enabling `APPLE_GMUX` has the side effect of also enabling `ACPI`.

zhovner, 2018-03-01:
You can look at `.config` file to see that `CONFIG_ACPI=y` is enabled.