cb22/macbook12-spi-driver #64: Vendor id and product id unset

Dunedan, 2018-04-08:
As @whot pointed out in https://github.com/Dunedan/mbp-2016-linux/issues/20#issuecomment-372188443, applespi currently doesn't set any vendor id or product id for the keyboard and the touchpad, preventing the automatic enabling of some features for Apple devices.

Here is how it looks in `/proc/bus/input/devices` for me:
```
I: Bus=001c Vendor=0000 Product=0000 Version=0000
N: Name="Apple SPI Keyboard"
P: Phys=applespi/input0
S: Sysfs=/devices/pci0000:00/0000:00:1e.3/pxa2xx-spi.3/spi_master/spi2/spi-APP000D:00/input/input18
U: Uniq=
H: Handlers=sysrq kbd leds tbkbd tbkbd event8
B: PROP=0
B: EV=120003
B: KEY=10000 0 0 0 6300001000 3800000000 710effd063c0001f ff7ffffffffffffe
B: LED=2

I: Bus=001c Vendor=0000 Product=0000 Version=0000
N: Name="Apple SPI Touchpad"
P: Phys=applespi/input1
S: Sysfs=/devices/pci0000:00/0000:00:1e.3/pxa2xx-spi.3/spi_master/spi2/spi-APP000D:00/input/input19
U: Uniq=
H: Handlers=mouse0 tbtpad tbtpad event9
B: PROP=5
B: EV=f
B: KEY=e520 10000 0 0 0 0
B: REL=3
B: ABS=27f800000000003
```

While the driver does currently use DMI for detecting the model and setting touchpad sizes accordingly, shouldn't or couldn't the SPI device provide such information itself somehow?

whot, 2018-04-08:
it should, but at least (and for now) it'd make sense to hardcode the apple vendor ID in there somewhere. Makes it pick up on a couple of behaviours.

roadrunner2, 2018-04-09:
Yes, the driver absolutely should put the proper ID's there, but so far we don't have them. I know of a single "info" query (from SPI traces) that possibly has that info in it, but I have no clue (yet) where inside that response the vendor and product ID lie.

As to hardcoding some ID, the question is, which ID and why? PCI? USB? Other? And why an Apple ID, since I think the keyboard and touchpad are made by Synaptics? And what happens if/when we finally manage to get the actual vendor id from SPI - do we switch to that and break existing stuff, or are we now stuck with a wrong id forever? (my guess is most likely the latter, since Linus is pretty adamant about kernel changes not breaking userspace) All in all, I'm not yet convinced that putting some other ID in there is a good idea.

Also, notice that most devices in `/proc/bus/input/devices` have a `Vendor=0000` (though they do have product ID's).

whot, 2018-04-09:
Well, that number is skewed, e.g. I have 7 "HDA Intel" devices for PCH and HDMI as well as the lid, sleep, power platform devices. Of the ones that are actual input devices (mice, keyboard, touchpads) only the keyboard has a garbage vendor ID because it's on the serial bus.

roadrunner2, 2018-04-22:
@Dunedan Sorry for the delay. I had that info query in my local stuff only, but I've now pushed it to my clone of this repo - if you build and load the `applespi` module from there, you'll see a bunch of 'applespi: info' logged which contain the response to this info query. For completeness sake, here is the response I see (MacBookPro13,3, US QWERTY keyboard):
```
40 d0 00 00 00 00 78 00 20 10 02 00 00 02 6e 00 05 01 09 02 a1 01 09 01 a1 00 05 09 19 01 29 03 
15 00 25 01 85 02 95 03 75 01 81 02 95 01 75 05 81 01 05 01 09 30 09 31 15 81 25 7f 75 08 95 02
81 06 95 04 75 08 81 01 c0 c0 05 0d 09 05 a1 01 06 00 ff 09 0c 15 00 26 ff 00 75 08 95 10 85 3f 
81 22 c0 06 00 ff 09 0c a1 01 06 00 ff 09 0c 15 00 26 ff 00 85 44 75 08 96 d7 06 81 00 c0 c0 fb
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 89 dc
```

Dunedan, 2018-04-29:
Here is the output of a MacBookPro13,2 with German keyboard layout. They differ, but the difference is pretty small:
```
40 d0 00 00 00 00 78 00 20 10 02 00 00 02 6e 00 05 01 09 02 a1 01 09 01 a1 00 05 09 19 01 29 03
15 00 25 01 85 02 95 03 75 01 81 02 95 01 75 05 81 01 05 01 09 30 09 31 15 81 25 7f 75 08 95 02
81 06 95 04 75 08 81 01 c0 c0 05 0d 09 05 a1 01 06 00 ff 09 0c 15 00 26 ff 00 75 08 95 10 85 3f
81 22 c0 06 00 ff 09 0c a1 01 06 00 ff 09 0c 15 00 26 ff 00 85 44 75 08 96 57 05 81 00 c0 c1 61
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 89 dc
```

roadrunner2, 2018-05-13:
So the difference is really just 2 bytes  (the other 2 bytes difference are the checksum): `d7 06` vs `57 05`. That could be related to either the keyboard layout or the touchpad size (other something else entirely) - we'll need dumps from a several more devices to narrow this down.

christophgysin, 2018-05-19:
Mine seems to be identical to @Dunedan but from a MBP13,1 with Finnish layout.
```
[ 3829.751986] applespi: info 40 d0 00 00 00 00 78 00 20 10 02 00 00 02 6e 00 05 01 09 02 a1 01 09 01 a1 00 05 09 19 01 29 03
[ 3829.751988] applespi: info 15 00 25 01 85 02 95 03 75 01 81 02 95 01 75 05 81 01 05 01 09 30 09 31 15 81 25 7f 75 08 95 02
[ 3829.751989] applespi: info 81 06 95 04 75 08 81 01 c0 c0 05 0d 09 05 a1 01 06 00 ff 09 0c 15 00 26 ff 00 75 08 95 10 85 3f
[ 3829.751990] applespi: info 81 22 c0 06 00 ff 09 0c a1 01 06 00 ff 09 0c 15 00 26 ff 00 85 44 75 08 96 57 05 81 00 c0 c1 61
[ 3829.751991] applespi: info 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[ 3829.751992] applespi: info 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[ 3829.751994] applespi: info 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[ 3829.751995] applespi: info 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 89 dc
```

peterychuang, 2018-05-19:
MacBookPro14,1, UK English keyboard

```
40 d0 00 00 00 00 78 00 20 10 02 00 00 02 6e 00 05 01 09 02 a1 01 09 01 a1 00 05 09 19 01 29 03
15 00 25 01 85 02 95 03 75 01 81 02 95 01 75 05 81 01 05 01 09 30 09 31 15 81 25 7f 75 08 95 02
81 06 95 04 75 08 81 01 c0 c0 05 0d 09 05 a1 01 06 00 ff 09 0c 15 00 26 ff 00 75 08 95 10 85 3f
81 22 c0 06 00 ff 09 0c a1 01 06 00 ff 09 0c 15 00 26 ff 00 85 44 75 08 96 57 05 81 00 c0 c1 61
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 89 dc
```

roadrunner2, 2018-05-20:
@christophgysin, @peterychuang: thanks for those. Yes, they're both identical to @Dunedan's output. I presume you both have/need the `iso_layout` parameter enabled?

peterychuang, 2018-05-20:
@roadrunner2 Indeed, ```iso_layout``` is set to ```1``` in my case. Not sure if that's relevant here, but I've also set ```fnmode``` to ```2```.

christophgysin, 2018-05-20:
Yes, I'm also using `iso_layout=1`.

So I guess we should be able to autodetect `iso_layout`?

Dunedan, 2018-05-20:
I'd really like to see an info query result from a MacBookPro13,3 or 14,3 with ISO keyboard to be sure it's related to ANSI vs. ISO and not to 13" vs. 15".

Also good would be a result from a 13" one with ANSI keyboard, as that'd be another interesting data point.

turenar, 2018-05-22:
MacBook Pro14,3 Japanese(JIS) Keyboard

```
[    4.920621] applespi: info 40 d0 00 00 00 00 78 00 20 10 02 00 00 02 6e 00 05 01 09 02 a1 01 09 01 a1 00 05 09 19 01 29 03
[    4.920622] applespi: info 15 00 25 01 85 02 95 03 75 01 81 02 95 01 75 05 81 01 05 01 09 30 09 31 15 81 25 7f 75 08 95 02
[    4.920622] applespi: info 81 06 95 04 75 08 81 01 c0 c0 05 0d 09 05 a1 01 06 00 ff 09 0c 15 00 26 ff 00 75 08 95 10 85 3f
[    4.920623] applespi: info 81 22 c0 06 00 ff 09 0c a1 01 06 00 ff 09 0c 15 00 26 ff 00 85 44 75 08 96 d7 06 81 00 c0 c0 fb
[    4.920623] applespi: info 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[    4.920623] applespi: info 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[    4.920624] applespi: info 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[    4.920624] applespi: info 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 89 dc
```

Dunedan, 2018-05-22:
Thanks @turenar, that's actually pretty interesting, as it shows the same values for an ANSI keyboard and a JIS keyboard. Right now it sounds likely to me that the differences in the output let us identify 13" and 15" models, but let's wait for the output of a MacBookPro13,3/14,3 with ISO keyboard to confirm.

Oh and the output of a 12" MacBook would be pretty interesting as well.

leifliddy, 2018-05-23:
Macbook 12" 2016 US keyboard

    40 d0 00 00 00 00 78 00 20 10 02 00 00 02 6e 00 05 01 09 02 a1 01 09 01 a1 00 05 09 19 01 29 03
    15 00 25 01 85 02 95 03 75 01 81 02 95 01 75 05 81 01 05 01 09 30 09 31 15 81 25 7f 75 08 95 02
    81 06 95 04 75 08 81 01 c0 c0 05 0d 09 05 a1 01 06 00 ff 09 0c 15 00 26 ff 00 75 08 95 10 85 3f
    81 22 c0 06 00 ff 09 0c a1 01 06 00 ff 09 0c 15 00 26 ff 00 85 44 75 08 96 17 04 81 00 c0 c1 52
    00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
    00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
    00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
    00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 89 dc

ClashTheBunny, 2018-05-23:
Macbook Pro 14,3, US Keyboard:
```
[    1.415388] applespi: info 40 d0 00 00 00 00 78 00 20 10 02 00 00 02 6e 00 05 01 09 02 a1 01 09 01 a1 00 05 09 19 01 29 03
[    1.415392] applespi: info 15 00 25 01 85 02 95 03 75 01 81 02 95 01 75 05 81 01 05 01 09 30 09 31 15 81 25 7f 75 08 95 02
[    1.415394] applespi: info 81 06 95 04 75 08 81 01 c0 c0 05 0d 09 05 a1 01 06 00 ff 09 0c 15 00 26 ff 00 75 08 95 10 85 3f
[    1.415396] applespi: info 81 22 c0 06 00 ff 09 0c a1 01 06 00 ff 09 0c 15 00 26 ff 00 85 44 75 08 96 d7 06 81 00 c0 c0 fb
[    1.415397] applespi: info 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[    1.415399] applespi: info 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[    1.415401] applespi: info 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[    1.415403] applespi: info 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 89 dc
```

Dunedan, 2018-05-24:
As there are even more models which support SPI (in addition to USB), the output of any MacBook Air or MacBook Pro released between Mid 2013 and 2015 would be interesting as well.

@newperson1746: I don't know how far you are getting a working driver for the MacBook Air, but if you can get the output of the info query it'd be awesome if you could post it here.

roadrunner2, 2018-05-25:
Thanks everyone so far! So yes, it looks like there's a touchpad model id in there - that'll certainly be helpful.

Regarding getting info from the MBA's: the reading of the response to a sent command is triggered by the receipt of the GPE; so as long as we're not getting those, we won't see the info command response.

melentye, 2018-07-08:
Another MacBookPro14,3 with US keyboard here, the output is identical to the one provided by @ClashTheBunny

roadrunner2, 2018-08-10:
I've pushed a change to [my clone](/roadrunner2/macbook12-spi-driver) that retrieves the id and 1) uses it to look up the touchpad info, and 2) populates the vendor and product id. If some of you could give it a whirl that would be great (basically it should work as before, and `grep -B1 -A8 Touchpad /proc/bus/input/devices` should show bus, vendor, and product id's of 001c, 06cb, and 06d7 or 0557 or 0417, respectively). TIA.

melentye, 2018-08-10:
@roadrunner2 I tried commit  cc3a895 from your fork and it doesn't seem to populate the bus, vendor and product id on MacBookPro14,3:

```
$ grep -B1 -A8 Touchpad /proc/bus/input/devices
I: Bus=001c Vendor=0000 Product=0000 Version=0000
N: Name="Apple SPI Touchpad"
P: Phys=applespi/input1
S: Sysfs=/devices/pci0000:00/0000:00:1e.3/pxa2xx-spi.5/spi_master/spi2/spi-APP000D:00/input/input6
U: Uniq=
H: Handlers=tbtpad tbtpad event6 mouse0 
B: PROP=5
B: EV=f
B: KEY=e520 10000 0 0 0 0
B: REL=3
```

melentye, 2018-08-11:
@roadrunner2 and just when I thought that I double-checked everything... The issue was with me forgetting to re-generate the initramfs which contains `applespi`.

It seems to work now:

```
$ grep -B1 -A8 Touchpad /proc/bus/input/devices
I: Bus=001c Vendor=06cb Product=06d7 Version=0000
N: Name="Apple SPI Touchpad"
P: Phys=applespi/input1
S: Sysfs=/devices/pci0000:00/0000:00:1e.3/pxa2xx-spi.5/spi_master/spi2/spi-APP000D:00/input/input6
U: Uniq=
H: Handlers=tbtpad tbtpad event6 mouse0 
B: PROP=5
B: EV=f
B: KEY=e520 10000 0 0 0 0
B: REL=3
```

sorry about the confusion earlier

roadrunner2, 2018-08-12:
@melentye Thanks for going back and getting it working!

Dunedan, 2018-08-25:
Works fine for me with Linux 4.18. Vendor id and product id for the touch pad are properly set. The ones for the keyboard are still missing, but that's expected as far as I can see.

roadrunner2, 2018-08-26:
Thanks for testing. Yes, don't have a keyboard "info" command yet, so no id's for that (and no automatic iso/non-iso layout determination :disappointed:).