cb22/macbook12-spi-driver #65: MacBookAir7,2 (Early 2015/2017) support

newperson1746, 2018-04-19:
Hi,
I have this very peculiar Macbook Air. It works in USB mode (applespi outputs something about device is in usb mode when booted normally via uefi grub) but using apple_set_os.efi to trick the efi into thinking its macOS results in the raw spi device being exposed. It uses AppleIntelLPSS and AppleIntelHSSPI just like the normal MacBook8,1. It has a Wildcat Point GSPI Controller with identical pci id's to the MacBook8,1:
```
$ sudo lspci -nnk | grep -i spi
00:15.4 Serial bus controller [0c80]: Intel Corporation Wildcat Point-LP Serial IO GSPI Controller #1 [8086:9ce6] (rev 03)
	Subsystem: Intel Corporation Wildcat Point-LP Serial IO GSPI Controller [8086:9ce6]
	Kernel driver in use: pxa2xx_spi_pci
	Kernel modules: spi_pxa2xx_pci

```
```
$ cat /etc/initramfs-tools/modules
# List of modules that you want to include in your initramfs.
# They will be loaded at boot time in the order below.
#
# Syntax:  module_name [args ...]
#
# You must run update-initramfs(8) to effect this change.
#
# Examples:
#
# raid1
# sd_mod

spi_pxa2xx_platform
spi_pxa2xx_pci
applespi

```
Some progress does occur when booting:
```
$ sudo dmesg | grep -i spi
[    0.000000] Command line: BOOT_IMAGE=/boot/vmlinuz-4.15.0-15-generic.efi.signed root=UUID=fe6baa00-1e02-4a13-bd0f-81652cd6024e ro acpi_osi=Darwin i915.modeset=1 pti=off applespi.debug=0x7 applespi.dyndbg=+p irqpoll
[    0.000000] Kernel command line: BOOT_IMAGE=/boot/vmlinuz-4.15.0-15-generic.efi.signed root=UUID=fe6baa00-1e02-4a13-bd0f-81652cd6024e ro acpi_osi=Darwin i915.modeset=1 pti=off applespi.debug=0x7 applespi.dyndbg=+p irqpoll
[    1.486690] applespi: loading out-of-tree module taints kernel.
[    1.488824] applespi: module verification failed: signature and/or required key missing - tainting kernel
[    5.079223] applespi: SPI settings: spi_cs_delay=10 reset_a2r_usec=10 reset_rec_usec=10
[    5.136477] input: Apple SPI Keyboard as /devices/pci0000:00/0000:00:15.4/pxa2xx-spi.0/spi_master/spi0/spi-APP000D:00/input/input10
[    5.136576] input: Apple SPI Touchpad as /devices/pci0000:00/0000:00:15.4/pxa2xx-spi.0/spi_master/spi0/spi-APP000D:00/input/input11
[    5.136723] applespi: spi-device probe done: spi-APP000D:00
```
```
$ xinput
⎡ Virtual core pointer                    	id=2	[master pointer  (3)]
⎜   ↳ Virtual core XTEST pointer              	id=4	[slave  pointer  (2)]
⎜   ↳ Mitsumi Electric Apple Optical USB Mouse	id=10	[slave  pointer  (2)]
⎜   ↳ Apple SPI Touchpad                      	id=14	[slave  pointer  (2)]
⎣ Virtual core keyboard                   	id=3	[master keyboard (2)]
    ↳ Virtual core XTEST keyboard             	id=5	[slave  keyboard (3)]
    ↳ Power Button                            	id=6	[slave  keyboard (3)]
    ↳ Video Bus                               	id=7	[slave  keyboard (3)]
    ↳ Power Button                            	id=8	[slave  keyboard (3)]
    ↳ Sleep Button                            	id=9	[slave  keyboard (3)]
    ↳ Logitech USB Keyboard                   	id=11	[slave  keyboard (3)]
    ↳ Logitech USB Keyboard                   	id=12	[slave  keyboard (3)]
    ↳ Apple SPI Keyboard                      	id=13	[slave  keyboard (3)]

```
But, they don't work on any part of linux boot or after, except on the grub screen. xinput test 13 and xinput test 14 do nothing. Amusingly, the keyboard backlight works without any additional work (it turns on during boot.) Using apple_set_os is beneficial since it unlocks some thunderbolt functionalities in linux that normally are reserved for macOS, and it would be nice to get it to work on one more device.

roadrunner2, 2018-04-19:
Given the dmesg and other info above, it looks like the SPI interface is properly enabled (and the USB interface disabled), so that's good. And especially the fact that the keyboard backlight is working is encouraging (does the caps-lock light also work, btw?).

Now to look at the keyboard and touchpad, can you enable debugging in the driver of the keyboard inputs:
```
echo 0x100 | sudo tee /sys/module/applespi/parameters/debug
```
And let me know what you see in dmesg when you type on the keyboard (there should basically be an applespi "Keyboard Event" packet dump for every key action) - this should let us know whether the driver is receiving and processing keyboard events or not.

Assuming keyboard events are being received, the next thing to do might be to take a look at libinput's info (assuming you're setup is using libinput): does running `libinput-debug-events` show any keyboard or touchpad activity?

newperson1746, 2018-04-20:
Alright, so I went ahead and ran the command exactly as you said. 
```
$ echo 0x100 | sudo tee /sys/module/applespi/parameters/debug
0x100
```
I then did
`$ dmesg -w
`
for a `tail -f` like experience (automatically updates as syslog does) and hit keys on the keyboard (I tried every single one.) No packet dumps appeared. After mashing a bunch of keys, here's all there is on `applespi`:
```
$ sudo dmesg | grep -i applespi
[    0.000000] Command line: BOOT_IMAGE=/boot/vmlinuz-4.15.0-15-generic.efi.signed root=UUID=fe6baa00-1e02-4a13-bd0f-81652cd6024e ro acpi_osi=Darwin i915.modeset=1 pti=off applespi.debug=0x7 applespi.dyndbg=+p
[    0.000000] Kernel command line: BOOT_IMAGE=/boot/vmlinuz-4.15.0-15-generic.efi.signed root=UUID=fe6baa00-1e02-4a13-bd0f-81652cd6024e ro acpi_osi=Darwin i915.modeset=1 pti=off applespi.debug=0x7 applespi.dyndbg=+p
[    1.566145] applespi: loading out-of-tree module taints kernel.
[    1.567789] applespi: module verification failed: signature and/or required key missing - tainting kernel
[    1.605546] applespi: SPI settings: spi_cs_delay=10 reset_a2r_usec=10 reset_rec_usec=10
[    1.666454] applespi: spi-device probe done: spi-APP000D:00

```
It appears like it is not receiving the packets. Interesting... Any thoughts?

newperson1746, 2018-04-20:
Yup:
```
$ echo 0xd00 | sudo tee /sys/module/applespi/parameters/debug
0xd00
```
```
$ dmesg | grep -i spi
[    0.000000] Command line: BOOT_IMAGE=/boot/vmlinuz-4.15.0-15-generic.efi.signed root=UUID=fe6baa00-1e02-4a13-bd0f-81652cd6024e ro acpi_osi=Darwin i915.modeset=1 pti=off applespi.debug=0x7 applespi.dyndbg=+p
[    0.000000] Kernel command line: BOOT_IMAGE=/boot/vmlinuz-4.15.0-15-generic.efi.signed root=UUID=fe6baa00-1e02-4a13-bd0f-81652cd6024e ro acpi_osi=Darwin i915.modeset=1 pti=off applespi.debug=0x7 applespi.dyndbg=+p
[    1.566145] applespi: loading out-of-tree module taints kernel.
[    1.567789] applespi: module verification failed: signature and/or required key missing - tainting kernel
[    1.605546] applespi: SPI settings: spi_cs_delay=10 reset_a2r_usec=10 reset_rec_usec=10
[    1.664089] input: Apple SPI Keyboard as /devices/pci0000:00/0000:00:15.4/pxa2xx-spi.0/spi_master/spi0/spi-APP000D:00/input/input4
[    1.665280] input: Apple SPI Touchpad as /devices/pci0000:00/0000:00:15.4/pxa2xx-spi.0/spi_master/spi0/spi-APP000D:00/input/input5
[    1.666454] applespi: spi-device probe done: spi-APP000D:00

```

roadrunner2, 2018-04-20:
Wow, no interrupts. Can you check your DSDT and see what GPE it declares for the keyboard/touchpad? (look for the `APP000D` device, and then the `_GPE` attribute in there) - alternatively, apply the following patch to the driver and check the dmesg output after (re)loading it:
```c
--- a/applespi.c
+++ b/applespi.c
@@ -1678,6 +1678,7 @@ static int applespi_probe(struct spi_device *spi)
                       acpi_format_exception(result));
                return -ENODEV;
        }
+       pr_info("GPE: %llx\n", gpe);
        applespi->gpe = (int)gpe;
 
        result = acpi_install_gpe_handler(NULL, applespi->gpe,
```
Next, given the GPE, show me the output of
```
cat /sys/firmware/acpi/interrupts/gpe<gpe-num-from-above-in-hex>
```
(on my system, the GPE is 0x17, btw).

newperson1746, 2018-04-20:
Yup, thankfully I have a hackintosh background, so I speak ACPI and AML now :)
I extracted it from the DSDT. My steps:
```
/sys/firmware/acpi/tables$ sudo apt install acpica-tools
/sys/firmware/acpi/tables$ sudo cp DSDT ~/Desktop
/sys/firmware/acpi/tables$ sudo chmod 777 DSDT
/sys/firmware/acpi/tables$ iasl -da -dl DSDT
/sys/firmware/acpi/tables$ grep -A 8 -B 8 -ira APP000d DSDT.dsl
                    Return (ConcatenateResTemplate (RBUF, DBUF))
                }
            }

            Scope (\_SB.PCI0.SPI1)
            {
                Device (SPIT)
                {
                    Name (_HID, EisaId ("APP000D"))  // _HID: Hardware ID
                    Name (_CID, "apple-spi-topcase")  // _CID: Compatible ID
                    Name (_DDN, "apple-spi-topcase")  // _DDN: DOS Device Name
                    Name (_GPE, 0x1C)  // _GPE: General Purpose Events
                    Name (_UID, 0x02)  // _UID: Unique ID
                    Name (_ADR, 0x00)  // _ADR: Address
                    Method (_PRW, 0, NotSerialized)  // _PRW: Power Resources for Wake
                    {
                        If (OSDW ())

```

roadrunner2, 2018-04-20:
What is the output of
```
cat /sys/firmware/acpi/interrupts/gpe1C
```

newperson1746, 2018-04-22:
Sorry about the delay! Here:
```
/sys/firmware/acpi/interrupts$ cat gpe1C
       1     STS enabled      unmasked

```

newperson1746, 2018-04-27:
I've been busy, sorry... The only change appears to be from:
```
/sys/firmware/acpi/interrupts$ cat gpe1C
       1     STS enabled      unmasked
```
To:
```
/sys/firmware/acpi/interrupts$ echo clear | sudo tee gpe1C
clear
/sys/firmware/acpi/interrupts$ cat gpe1C
       1         enabled      unmasked
```
dmesg shows no change, xinput still doesn't show any data from either SPI device, and dmesg shows no change. This is really a head-scratcher. I can patch the ACPI tables if necessary, I've already done so on an older Mac on Ubuntu to fix thunderbolt (with the _OSI call patching to report Darwin every time.) Knowing that the interrupts are over GPIO, maybe we need to enable them via a special method in the ACPI tables? FYI the MacBookAir6,2 tables in the github link are identical to mine, so that's a good way to look at the DSDT if we need to. Sorry if I'm dragging this too far, it's a lot, I know, and I appreciate your work.

l1k, 2018-05-12:
@newperson1746: That code comment that the MBP12,1 is the only machine with both, USB and SPI, was written by me and is clearly erroneous. I thought that's the case but I recently came across a <a href="https://www.youtube.com/watch?v=geGmC1xI4zo">video</a> by Louis Rossmann which proves that MacBook Airs also had this. Feel free to submit a pull request rectifying the incorrect code comment.

Could you provide a little more detail which Thunderbolt functionalities are enabled by apple_set_os?

l1k, 2018-05-12:
The MBA6,2 DSDT linked above seems to be missing the device properties and _CRS resources for the SPIT device (clock etc), I guess we need to define defaults for this case. :-(

roadrunner2, 2018-05-12:
@newperson1746 Sorry, I missed your edits. Regarding the code and the `applespi_default_info`: those values are "just" the range of x/y values the touchpad reports - while we'll quite possibly need to figure out those for the macbook air (though they well might be the same as those for the macbook's), we don't need to worry about them until we get the interrupt/gpe issue worked out.

Regarding the repo's: my clone is mainly for two things, 1) the touchbar driver, and 2) things that aren't ready/meant to be pulled into the main repo. So please submit your PR to cb22's repo.

l1k, 2018-05-13:
@newperson1746: Could you add a debug printk to applespi's `->probe` routine to output the `max_speed_hz` of the SPI device? Since there are no device properties in the ACPI namespace on the MBA6, I'd expect `max_speed_hz` to be zero. However <a href="https://elixir.bootlin.com/linux/latest/source/drivers/spi/spi.c#L1805">`acpi_register_spi_device()`</a> bails out if it's zero. So why is the driver probing at all? Maybe on the MBA7 those device properties are present? Could you upload an acpidump of your MBA7 somewhere?

If you're talking to the device with the wrong clock, it wouldn't be surprising if the GPE never fires.

As for Thunderbolt, since commit torvalds/linux@7bc5a2b it should no longer be necessary to pass `acpi_osi=Darwin` on Macs as that's the default. Quite the opposite, one has to turn the Darwin OSI string  *off* if one doesn't want to use Thunderbolt. Also, the ACPI OSI protocol is a separate thing from the EFI apple_set_os protocol. I'm not aware that the latter has any influence on Thunderbolt. However I agree that it would be good to have MBA support in the applespi driver.

newperson1746, 2018-05-13:
Here are all the ACPI tables, including the dynamic ones (from /sys/firmware/acpi/tables and /sys/firmware/acpi/tables/dynamic) and disassembled in the ./disassembled folder using iasl -da -dl. I am not too great at writing C, so could you give me a hand in adding printk? Sorry to have gone this far in problem-reporting without having real C experience... 
[AllACPI.zip](https://github.com/cb22/macbook12-spi-driver/files/1998579/AllACPI.zip)

I think I confused apple_set_os and acpi_osi=Darwin. You are right about not needing to pass OSI anymore. apple_set_os basically enables the discrete gpu in dual-gpu macbook pros, enables the raw battery protocol (instead of the acpi-compliant one, but it works in linux nonetheless) and switches the keyboard + trackpad (bcm5974) into SPI mode. Also note that the keyboard backlight turns on and is controllable as an led with range 0-255 in /sys/class/leds/smc::kbd_backlight, even without applespi loaded (applesmc must control the backlight through the SMC, and not the actual keyboard interface.) 

Well, it's great to see great people working on a great driver. Thank you for your support.

roadrunner2, 2018-05-14:
Thanks for the DSDT - that looks good, i.e. it contains the expected _DSM method for the spi device, from which we extract various parameters for the SPI driver. And that explains why the applespi driver loads fine.

It's not really necessary now, but just for kicks, here is basically the patch @l1k was asking for:
```patch
--- a/applespi.c
+++ b/applespi.c
@@ -1638,6 +1638,16 @@ static int applespi_probe(struct spi_device *spi)
        if (result)
                return result;
 
+       {
+               struct acpi_device *adev = ACPI_COMPANION(&applespi->spi->dev);
+               const union acpi_object *o;
+
+               if (!acpi_dev_get_property(adev, "spiSclkPeriod", ACPI_TYPE_BUFFER, &o))
+                       pr_info("spiSclkPeriod: %llu\n", *(u64 *)o->buffer.pointer);
+               else
+                       pr_warn("Property spiSclkPeriod not found\n");
+       }
+
        /* set up touchpad dimensions */
        applespi->tp_info = *(struct applespi_tp_info *)
                        dmi_first_match(applespi_touchpad_infos)->driver_data;
```
Compiling and loading the new driver should result in a 'spiSclkPeriod: 125' message in dmesg.

newperson1746, 2018-08-19:
Hi everyone,
It's been a while. I wanted to see how everything was in the current state of the driver. I am using the exact same laptop with the exact same Ubuntu install with the 4.18.3 kernel and applespi as a dkms module, exactly like it was a couple months ago. I want to help out (now that I have some time!) 

The log from a boot with the spi controller enabled with apple_set_os:
[    1.487001] applespi: loading out-of-tree module taints kernel.
[    1.488326] applespi: module verification failed: signature and/or required key missing - tainting kernel
[    1.510569] applespi: SPI settings: spi_cs_delay=10 reset_a2r_usec=10 reset_rec_usec=10
[    1.581979] applespi: spi-device probe done: spi-APP000D:00
[    1.784082] applespi spi-APP000D:00: SPI transfer timed out
[    1.784095] applespi: --- Touchpad Initialization ------------------------
[    1.784097] applespi: write  40 02 00 00 00 00 0c 00 52 02 00 00 02 00 02 00 02 01 7b 11 00 00 00 00 00 00 00 00 00 00 00 00
[    1.784098] applespi: write  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[    1.784099] applespi: write  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[    1.784100] applespi: write  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[    1.784101] applespi: write  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[    1.784102] applespi: write  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[    1.784103] applespi: write  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[    1.784104] applespi: write  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 23 ab
[    1.784105] applespi: status 00 00 00 00
[    1.784106] applespi: Error writing to device: -110
[    1.988043] applespi spi-APP000D:00: SPI transfer timed out
[    1.988064] applespi: Error reading from device: -110
[    2.192022] applespi spi-APP000D:00: SPI transfer timed out
[    2.192029] applespi: Error reading from device: -110
[    2.396101] applespi spi-APP000D:00: SPI transfer timed out
[    2.396108] applespi: Error reading from device: -110
[    2.600103] applespi spi-APP000D:00: SPI transfer timed out
[    2.600109] applespi: Error reading from device: -110
[    2.804101] applespi spi-APP000D:00: SPI transfer timed out
[    2.804108] applespi: Error reading from device: -110
[    3.008118] applespi spi-APP000D:00: SPI transfer timed out
[    3.008126] applespi: Error reading from device: -110
[    3.212018] applespi spi-APP000D:00: SPI transfer timed out
[    3.212025] applespi: Error reading from device: -110
[    3.416101] applespi spi-APP000D:00: SPI transfer timed out
[    3.416108] applespi: Error reading from device: -110
[    3.620039] applespi spi-APP000D:00: SPI transfer timed out
[    3.621464] applespi: Error reading from device: -110
[    3.824048] applespi spi-APP000D:00: SPI transfer timed out
[    3.829724] applespi: Error reading from device: -110
[    4.040031] applespi spi-APP000D:00: SPI transfer timed out
[    4.041075] applespi: Error reading from device: -110
[    4.244018] applespi spi-APP000D:00: SPI transfer timed out
[    4.244026] applespi: Error reading from device: -110
[    4.448019] applespi spi-APP000D:00: SPI transfer timed out
[    4.449184] applespi: Error reading from device: -110
[    4.652014] applespi spi-APP000D:00: SPI transfer timed out
[    4.653438] applespi: Error reading from device: -110
[    4.856024] applespi spi-APP000D:00: SPI transfer timed out
[    4.856030] applespi: Error reading from device: -110
[    5.060018] applespi spi-APP000D:00: SPI transfer timed out
[    5.061506] applespi: Error reading from device: -110
[    5.264018] applespi spi-APP000D:00: SPI transfer timed out
[    5.265476] applespi: Error reading from device: -110
[    5.468268] applespi spi-APP000D:00: SPI transfer timed out
[    5.468277] applespi: Error reading from device: -110
[    5.672020] applespi spi-APP000D:00: SPI transfer timed out
[    5.672029] applespi: Error reading from device: -110
[    5.876024] applespi spi-APP000D:00: SPI transfer timed out
[    5.877295] applespi: Error reading from device: -110
[    6.080029] applespi spi-APP000D:00: SPI transfer timed out
[    6.080037] applespi: Error reading from device: -110
[    6.284019] applespi spi-APP000D:00: SPI transfer timed out
[    6.284027] applespi: Error reading from device: -110
[    6.488047] applespi spi-APP000D:00: SPI transfer timed out
[    6.489105] applespi: Error reading from device: -110
[    6.692020] applespi spi-APP000D:00: SPI transfer timed out
[    6.692031] applespi: Error reading from device: -110
[    6.896048] applespi spi-APP000D:00: SPI transfer timed out
[    6.896058] applespi: Error reading from device: -110
[... 112 lines cut ...]
[   18.544024] applespi spi-APP000D:00: SPI transfer timed out
[   18.544036] applespi: Error reading from device: -110
[   18.748041] applespi spi-APP000D:00: SPI transfer timed out
[   18.748053] applespi: Error reading from device: -110
[   18.952033] applespi spi-APP000D:00: SPI transfer timed out
[   18.952044] applespi: Error reading from device: -110
[   19.156032] applespi spi-APP000D:00: SPI transfer timed out
[   19.156044] applespi: --- Caps-Lock Command ------------------------
[... cut ...]
[   19.156055] applespi: write  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 66 6a
[   19.156056] applespi: status 00 00 00 00
[   19.156058] applespi: Error writing to device: -110
[   19.360030] applespi spi-APP000D:00: SPI transfer timed out
[   19.360042] applespi: Error reading from device: -110
[   19.564020] applespi spi-APP000D:00: SPI transfer timed out
[   19.564032] applespi: Error reading from device: -110
[   19.768024] applespi spi-APP000D:00: SPI transfer timed out

The Caps Lock messages then loop along with the -110 errors. How should I begin?

roadrunner2, 2018-08-26:
Sorry for the delay. First of all: what version exactly of applespi are you running?

Next, either something's changed or I misread things last time, because your latest log seems to indicate the GPE is being received and handled - great! Though to confirm, could you enable a few more debug bits? I'm guessing you're booting with something like `applespi.debug=0xf` on the kernel command line? If so, then please try `applespi.debug=0xff` - this should also explicitly log whenever our GPE handler is invoked.

Finally, the problem now appears to be that the SPI device (keyboard/touchpad) is not responding. I see several potential reasons for this:
- the clock frequency is too high (though I think we are getting it correctly from the DSDT)
- additional delays are necessary between the different transfers (we can insert/increase them and see if it helps)
- the keyboard/touchpad SPI interface is not properly enabled (but we do call SIEN ACPI method for this)

To double check the SIEN call, can you apply the following patch and then send me the `dmesg` output after reloading/rebooting?
```patch
--- a/applespi.c
+++ b/applespi.c
@@ -799,6 +799,7 @@ static int applespi_enable_spi(struct applespi_data *applespi)
 
        /* check if SPI is already enabled, so we can skip the delay below */
        result = acpi_evaluate_integer(applespi->sist, NULL, NULL, &spi_status);
+       pr_info("SIST before: res=%d sts=%llu\n", result, spi_status);
        if (ACPI_SUCCESS(result) && spi_status)
                return 0;
 
@@ -809,6 +810,9 @@ static int applespi_enable_spi(struct applespi_data *applespi)
                return -ENODEV;
        }
 
+       result = acpi_evaluate_integer(applespi->sist, NULL, NULL, &spi_status);
+       pr_info("SIST after: res=%d sts=%llu\n", result, spi_status);
+
        /*
         * Allow the SPI interface to come up before returning. Without this
         * delay, the SPI commands to enable multitouch mode may not reach
```