cb22/macbook12-spi-driver #69: Explanation of how to install kernel module to get keyboard working

wolfsta, 2018-10-12:
Hi,

I have a clean install of ubuntu, and am not sure exactly on how to get the second module.  lsmod | grep spi comes back as having 'spi_pxa2xx_platform' installed.

How do I install the second pci one?

Thanks

roadrunner2, 2018-10-14:
You don't mention what model of MacBook(Pro) you're running, or what version of Ubuntu. But from the above I'm guessing you've got a MacBook8,1 (2015) then and Ubuntu 18.04? If so, you should be able to load the module with `modprobe spi_pxa2xx_pci`.

wolfsta, 2018-10-15:
Hi, thanks for the response, i have a 2016 MBP and ubuntu 18.04,  i managed to get the keyboard and trackpad running following the step by step keyboard section here: https://github.com/chisNaN/ubuntu-on-macbook12.

Next I just need some time to get the touchbar working partially for escape key etc.

Thanks

xo1, 2018-10-27:
@roadrunner2 Hi, I have been following your instructions and am getting stuck at the very last bit.

Machine: **Macbook Pro 2017** (with touchbar)
uname -a: **Linux kali 4.12.0-kali1-amd64 #1 SMP Debian 4.12.6-1kali6 (2017-08-30) x86_64 GNU/Linux**
Folders in /usr/src:
**applespi-0.1                      
linux-kbuild-4.18
linux-headers-4.18.0-kali2-amd64
volatility-tools
linux-headers-4.18.0-kali2-common**

So far I am unable to get the keyboard and touchpad to work.

When I run: **dkms install -m applespi -v 0.1**

I see: 

**Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area.......................**

But this goes on forever and does not stop. Sorry for being a newbie, I have spent days trying to get this to work and this is furthest I have got so far.

Hope you can help. I am happy to make a donation for your support, I am just a student, thank you.

roadrunner2, 2018-10-28:
@xo1 You mean the dots after "cleaning build area" just keep on going, and nothing else ever shows up? That is odd - something is obviously stuck. Try seeing what process that is, e.g. by opening another terminal, searching for the dkms process (e.g. `ps -ef | grep dkms`) and then running `pstree <pid-of-dkms-process>` to see what could be hanging.

MakGulati, 2019-04-19:
> Check out [my clone of this repo](/roadrunner2/macbook12-spi-driver) for the touchbar.
> 
> Edit: oh, sorry, I see those instructions reference my repo already. In that case you'll just need to take into account that the `appletb` driver now depends on the `industrialio_triggered_buffer` module being loaded - it's easiest if you just use dkms to build the modules (see the "DKMS module" section in the README) and **then use `modprobe appletb` to load the touchbar driver.**
could you explain this step.
thank you so much