cb22/macbook12-spi-driver #72: Palms often touch-click on trackpad while typing on keyboard

gygias, 2019-01-10:
On my MacBook14,1 and Ubuntu 18.04, often while typing if the palms of my hands graze the large trackpad, it clicks, losing focus on where I was typing. I like using the touch-to-click setting anyway, so I'm hoping to find a way to filter out this input, whether it's determined by "I'm currently typing" or some kind of area of surface touching the trackpad.

Thank you for this excellent driver by the way.

gygias, 2019-01-10:
*MacBookPro14,1

roadrunner2, 2019-01-11:
Yes, this even more of an issue on the 15" models. However, palm detection is (mostly) up to the higher level drivers, in particular libinput or Xorg's synaptics driver. In particular libinput's palm detection does not work very well for these huge touchpads (it's based largely on exclusion zones and touch size - see https://wayland.freedesktop.org/libinput/doc/latest/palm-detection.html). I use the following overrides in `/etc/libinput/local-overrides.quirks` (for libinput 1.12) that help a bit:
```ini
[MacBook(Pro) SPI Touchpads]
MatchName=*Apple SPI Touchpad*
ModelAppleTouchpad=1
AttrKeyboardIntegration=internal
AttrTouchSizeRange=200:150
AttrPalmSizeThreshold=1100
```

Having said that, libinput does recognize `MT_TOOL_PALM` if set by the touchpad driver, and I've seen some indication that the touchbar data we receive contains some palm detection info (in some of the 'unknown' fields in the `struct touchpad_protocol` data), but I've yet to make proper sense of it; and even if I did and we start setting `MT_TOOL_PALM`, it's not clear to me that this signal would be any better than the current stuff done by libinput (or the synaptics driver).

Lastly, I have played with trying to improve libinput's palm detection, capturing traces of palm and finger touches and trying to come up with an algorithm that could reliably distinguish between the two, but so far haven't had any luck. I'm thinking that we might need to throw some machine learning at it, something which I've heard whispers that Apple does.

jottr, 2024-03-19:
Has there been any development regarding palm detection? 

>  I've seen some indication that the touchbar data we receive contains some palm detection info (in some of the 'unknown' fields in the struct touchpad_protocol data), but I've yet to make proper sense of it

What is missing so that we are able to understand the labels emitted by the touchpad firmware?