cb22/macbook12-spi-driver #76: Can install applespi v0.1

HassanDuhaini, 2019-12-24:
I get this error when running the installation command:
----------------------------------------------------------------------------------------------------------------------------------
Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
make -j8 KERNELRELEASE=5.4.0-050400-lowlatency...(bad exit status: 2)
ERROR (dkms apport): binary package for applespi: 0.1 not found
Error! Bad return status for module build on kernel: 5.4.0-050400-lowlatency (x86_64)
Consult /var/lib/dkms/applespi/0.1/build/make.log for more information.
-------------------------------------------------------------------------------------------------------------------------------

I am running Ubuntu 18 on Macbook pro 15in 2017 model.

roadrunner2, 2019-12-24:
As of kernel 5.3 this module is already included so you might not need to build it yourself at all - see if CONFIG_KEYBOARD_APPLESPI is enabled in your /boot/config-... .

If it's not enabled, then please get the latest driver from https://github.com/roadrunner2/macbook12-spi-driver/ as this repo here is somewhat out of date.

arno01, 2019-12-30:
Just a side note: while applespi has been included to Linux 5.3, the touch bar driver wasn't.. so we still need to compile the driver from https://github.com/roadrunner2/macbook12-spi-driver/ repo.

roadrunner2, 2020-02-24:
@HassanDuhaini: Can this issue be closed?