cb22/macbook12-spi-driver #77: Is this driver valid for MacBookPro15,4?

abhisek176, 2019-12-29:


arno01, 2019-12-30:
Hi @abhisek176 , I've updated the article https://nixaid.com/linux-on-macbookpro/
Could you please try booting the https://files.nixaid.com/ubuntu4mac.iso ? (``sha256: a77086a4aadbfd5b29e86232a9e24aba0f3a963af064e3c961adae8e9f8d2682``)

I've added the ``evdev:name:Apple SPI Touchpad:dmi:*:svnAppleInc.:pnMacBookPro15,4:*`` line to ``/etc/udev/hwdb.d/61-evdev-local.hwdb`` file there. Cannot promise that it will help, but let's see.

Let me know whether the keyboard, touch pad, touch bar are working.

abhisek176, 2019-12-30:
Hi @arno01, I tried the new ISO. The bootable flash drive did not show any sign of trackpad or keyboard movement. I tried installing but the installation got failed with an error [https://drive.google.com/open?id=1_LlxRSGJZbKVX_pMErFjJQJs3OEj_AnR](url)

arno01, 2019-12-30:
@abhisek176 regarding the installer error, I have replied you via email since it's an off-topic.

> The bootable flash drive did not show any sign of trackpad or keyboard movement.
I would presume the Linux kernel isn't recognizing some of the new MBP's 2019 hardware properly.
In this case there is no need to proceed installing it

Hook an external keyboard and you will be able to run some commands which might in figuring out why the driver isn't working.

Please provide us with the output of these commands:

- ``lspci -nnk``
- ``dmesg -T -l err``
- ``cat /proc/interrupts``
- ``dmesg -T | grep -i spi``
- ``xinput``

abhisek176, 2019-12-30:
Hi @arno01: here are the outputs https://drive.google.com/open?id=1QtABl_PQE-dzv7O24glU9Ct7fFxGR5OJ. please have a look. kindly let me know if you need anything.

arno01, 2019-12-30:
Abhisek was also sending me his syslog file, here is his HW:

```
DMI: Apple Inc. MacBookPro15,4/Mac-REDACTED, BIOS 1037.60.58.0.0 (iBridge: 17.16.12551.0.0,0) 11/14/2019
```

And he's got these errors as well:

```
Dec 30 12:41:33 ubuntu kernel: [    3.460140] applespi: loading out-of-tree module taints kernel.
Dec 30 12:41:33 ubuntu kernel: [   12.735839] applesmc: send_byte(0x10, 0x0304) fail: 0xff
Dec 30 12:41:33 ubuntu kernel: [   12.735840] applesmc: #KEY: read arg fail
...
Dec 30 12:42:49 ubuntu kernel: [   87.903816] applesmc: #KEY: read arg fail
Dec 30 12:42:49 ubuntu kernel: [   87.960613] applesmc: driver init failed (ret=-5)!
```

Let's see if @roadrunner2 & @cb22 will have some more clues based on the logs provided by @abhisek176 .

arno01, 2019-12-30:
@abhisek176 I've just found something that you might want to try.

There is a mbp15 branch https://github.com/roadrunner2/macbook12-spi-driver/tree/mbp15 - Work in progress driver for the touchbar and ambient-light-sensor on 2019 MacBook Pro's.

https://github.com/MCMrARM/mbp2018-bridge-drv - 
A driver for MacBook models 2018 and newer, which makes the keyboard, mouse and audio output work.

roadrunner2, 2019-12-31:
@abhisek176 The MacBook Pro 15,* models do not expose an SPI based keyboard and trackpad, but instead a USB based one, i.e the driver here is only for 13,* and 14.* models. The links @arno01 provides above are correct: you need @MCMrARM's driver for the keyboard and trackpad, and my driver for the touchbar and ALS.

abhisek176, 2020-01-02:
Hi @roadrunner2 . I used @MCMrARM driver and it worked. But I having problem with the touchpad clicks and taps.

roadrunner2, 2020-02-24:
@abhisek176: can this issue be closed?

abhisek176, 2020-02-24:
Is there any further update on this? Because I still am unable to find a workaround on the touchpad clicks and taps :(

roadrunner2, 2020-02-24:
Two things: 1) you never actually described what your touchpad issues are exactly, and 2) whatever they are, they aren't related to this driver since you're using a different driver.

abhisek176, 2020-02-24:
The touchpad click (right) and taps not working. Thanks!

closed 2020-02-24 by abhisek176

roadrunner2, 2020-02-24:
It sounds like your issue is most likely with the libinput settings - try running `libinput debug-events --enable-tap` and see if that shows taps working. And take a look at your desktop-environment's trackpad settings and [libinput quirks](https://wayland.freedesktop.org/libinput/doc/latest/device-quirks.html).

abhisek176, 2020-02-25:
I tried it. Shows below:
-event3   DEVICE_ADDED     Power Button                      seat0 default group1  cap:k
-event4   DEVICE_ADDED     Video Bus                         seat0 default group2  cap:k
-event1   DEVICE_ADDED     Power Button                      seat0 default group3  cap:k
-event0   DEVICE_ADDED     Lid Switch                        seat0 default group4  cap:S
-event2   DEVICE_ADDED     Sleep Button                      seat0 default group5  cap:k
-event5   DEVICE_ADDED     USB Optical Mouse                 seat0 default group6  cap:p left scroll-nat scroll-button
-event13  DEVICE_ADDED     Apple Headset                     seat0 default group7  cap:k
-event14  DEVICE_ADDED     Apple Inc. Apple Internal Keyboard / Trackpad Keyboard seat0 default group8  cap:k
-event15  DEVICE_ADDED     Apple Inc. Apple Internal Keyboard / Trackpad Consumer Control seat0 default group8  cap:k
-event16  DEVICE_ADDED     Apple Inc. Apple Internal Keyboard / Trackpad Mouse seat0 default group8  cap:p left scroll-nat scroll-button
-event17  DEVICE_ADDED     Apple Inc. Touch Bar Display      seat0 default group9  cap:k
-event7   DEVICE_ADDED     Apple T2 Audio Codec Output       seat0 default group10 cap:
-event10  DEVICE_ADDED     HDA Intel PCH HDMI/DP,pcm=8       seat0 default group10 cap:
-event11  DEVICE_ADDED     HDA Intel PCH HDMI/DP,pcm=9       seat0 default group10 cap:
-event12  DEVICE_ADDED     HDA Intel PCH HDMI/DP,pcm=10      seat0 default

roadrunner2, 2020-02-26:
The question is what does it show when you tap and click, with one or multiple fingers? In my case I see BTN_LEFT, BTN_RIGHT, BTN_MIDDLE events when tapping with one, two, and three fingers, but always BTN_LEFT with clicking with any number of fingers (though the final result, namely that of getting left, right, and middle click behaviour, is correct - I'm not enough of a libinput expert to know why this difference exists for actual clicks).