cb22/macbook12-spi-driver #79: Touchbar doesn't work on MacbookPro 15,2

mfvianna, 2020-04-12:
I've made almost everything to work, including keyboard, sensors, sound and WiFi with the kernel patches from [https://github.com/aunali1/linux-mbp-arch](https://github.com/aunali1/linux-mbp-arch) and the BCE driver from [https://github.com/MCMrARM/mbp2018-bridge-drv](https://github.com/MCMrARM/mbp2018-bridge-drv).
No need to use the SPI driver from [https://github.com/roadrunner2/macbook12-spi-driver](https://github.com/roadrunner2/macbook12-spi-driver) to make any of this to work (even for the keyboard or touchpad), the BCE driver suffices.

I've tried, though, to use the SPI driver specifically for the touch bar. The modules compile and load fine, but nothing happens.

I'd like to support this. Currently I don't have much time to code, but I can test and provide logs, system info collected from lshw, lsusb, dmesg, etc. and of course do some local compilations and test or make minor code adjustments.

roadrunner2, 2020-04-13:
The SPI driver is for the keyboard and trackpad only - it has nothing to do with the touchbar (which sits on the USB bus). To get the touchbar driver for MBP15,*, go to my clone of this repo (https://github.com/roadrunner2/macbook12-spi-driver) and check out the mbp15 branch.

mfvianna, 2020-04-13:
> The SPI driver is for the keyboard and trackpad only - it has nothing to do with the touchbar (which sits on the USB bus). To get the touchbar driver for MBP15,*, go to my clone of this repo (https://github.com/roadrunner2/macbook12-spi-driver) and check out the mbp15 branch.

Thank you so much!!!

Sorry for the confusion.  I mentioned spi driver because of the name of the project "macbook12-**spi**-driver", my bad, sorry for the ignorance.  What I really meant was the ibridge driver.

roadrunner2, 2020-04-13:
> Is it possible to configure the touchbar keys?

Somewhat: there's very limited choices in the basic driver (one of 3 layouts, with the fn key switching between them: esc-only, esc+function-keys, esc+special-keys). But some work has been done on getting the full fancy touchbar functionality working here (look near the bottom of the discussion): Dunedan/mbp-2016-linux#110

jbg, 2020-04-13:
I have the exact same configuration as the OP in this issue; kernel from https://github.com/aunali1/linux-mbp-arch, `apple-bce` module from https://github.com/MCMrARM/mbp2018-bridge-drv and iBridge/touchbar/ALS driver from https://github.com/roadrunner2/macbook12-spi-driver/tree/mbp15. The touchbar was working until a few days ago but now doesn't work. I've tried reverting all recent package updates and it still doesn't work. I found roadrunner2/macbook12-spi-driver#42 and libimobiledevice/usbmuxd#138 so patched the usbmuxd udev rules to only apply to removable devices, but the touchbar still doesn't work. What should I do to debug this?

(Apologies if my issue turns out to be different than OP's - I have the exact problem in the issue title and same configuration, so didn't want to open a duplicate issue.)

roadrunner2, 2020-04-13:
@jbg Is the `apple-ibridge` module loaded too? Can you show me the output of `lsusb -tv`? Also, are there any errors in the kernel log (dmesg)?

jbg, 2020-04-14:
Yes, `apple-ibridge` is loaded. There are no errors in dmesg, nor any log lines at all containing the name of any of the modules; the only log lines containing `apple` are from apple-bce setting up audio.

```
# lsusb -tv
/:  Bus 07.Port 1: Dev 1, Class=root_hub, Driver=/12p, 480M
    ID 1d6b:0002 Linux Foundation 2.0 root hub
    |__ Port 1: Dev 2, If 1, Class=CDC Data, Driver=cdc_ncm, 480M
        ID 05ac:8233 Apple, Inc.
    |__ Port 1: Dev 2, If 0, Class=Communications, Driver=cdc_ncm, 480M
        ID 05ac:8233 Apple, Inc.
    |__ Port 2: Dev 3, If 1, Class=Video, Driver=uvcvideo, 480M
        ID 05ac:8514 Apple, Inc.
    |__ Port 2: Dev 3, If 0, Class=Video, Driver=uvcvideo, 480M
        ID 05ac:8514 Apple, Inc.
    |__ Port 3: Dev 4, If 0, Class=Human Interface Device, Driver=usbhid, 480M
        ID 05ac:8262 Apple, Inc.
    |__ Port 4: Dev 5, If 0, Class=Human Interface Device, Driver=usbhid, 480M
        ID 05ac:8103 Apple, Inc.
    |__ Port 5: Dev 6, If 3, Class=Human Interface Device, Driver=usbhid, 480M
        ID 05ac:027b Apple, Inc.
    |__ Port 5: Dev 6, If 1, Class=Human Interface Device, Driver=usbhid, 480M
        ID 05ac:027b Apple, Inc.
    |__ Port 5: Dev 6, If 4, Class=Human Interface Device, Driver=usbhid, 480M
        ID 05ac:027b Apple, Inc.
    |__ Port 5: Dev 6, If 2, Class=Human Interface Device, Driver=bcm5974, 480M
        ID 05ac:027b Apple, Inc.
    |__ Port 5: Dev 6, If 0, Class=Human Interface Device, Driver=usbhid, 480M
        ID 05ac:027b Apple, Inc.
    |__ Port 6: Dev 7, If 0, Class=Human Interface Device, Driver=usbhid, 480M
        ID 05ac:8302 Apple, Inc.
    |__ Port 7: Dev 8, If 0, Class=Human Interface Device, Driver=usbhid, 480M
        ID 05ac:8102 Apple, Inc.
/:  Bus 06.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/2p, 10000M
    ID 1d6b:0003 Linux Foundation 3.0 root hub
/:  Bus 05.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/2p, 480M
    ID 1d6b:0002 Linux Foundation 2.0 root hub
/:  Bus 04.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/2p, 10000M
    ID 1d6b:0003 Linux Foundation 3.0 root hub
/:  Bus 03.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/2p, 480M
    ID 1d6b:0002 Linux Foundation 2.0 root hub
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/6p, 10000M
    ID 1d6b:0003 Linux Foundation 3.0 root hub
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/12p, 480M
    ID 1d6b:0002 Linux Foundation 2.0 root hub
```

I misspoke in my earlier post; the camera does work, but the ALS and touch bar do not.